reaper7/SDM_Energy_Meter #2: Adaption to sdm630??

beireken, 2016-10-26:
Hello,
Your library seems to be working fine. Thank you for sharing!
I however am looking to also read from a 3phase sdm630.
Can this library be easily adapted to add the 3phase registers?

Regards,
Bart

reaper7, 2016-10-26:
You are try read sdm630 with this lib?
If You can read parameters for first phase, then rest will be possible too.
Look inside registers for sdm630:
http://www.photovoltaikforum.com/download/file.php?id=58606

sdm630 registers for first phase are identical with sdm220...

so, You can add definitions for rest of registers, something like this:
#define SDM_VOLTAGE1                 0x0000
#define SDM_VOLTAGE2                 0x0002
#define SDM_VOLTAGE3                 0x0004
#define SDM_CURRENT1                 0x0006
#define SDM_CURRENT1                 0x0008
#define SDM_CURRENT1                 0x000A
etc...
+ additional registers, readable only from sdm630

beireken, 2016-10-26:
OK, thx.
I did not think it would be that easy...

So i adapted the library to this:

`#define SDM120C_VOLTAGE                 0x0000                                      //V
#define SDM120C_CURRENT                 0x0006                                      //A
#define SDM120C_POWER                   0x000C                                      //W
#define SDM120C_ACTIVE_APPARENT_POWER   0x0012                                      //VA
#define SDM120C_REACTIVE_APPARENT_POWER 0x0018                                      //VAR
#define SDM120C_POWER_FACTOR            0x001E                                      //
#define SDM120C_PHASE_ANGLE             0x0024                                      //DEGREE
#define SDM120C_FREQUENCY               0x0046                                      //Hz
#define SDM120C_IMPORT_ACTIVE_ENERGY    0x0048                                      //Wh
#define SDM120C_EXPORT_ACTIVE_ENERGY    0x004A                                      //Wh
#define SDM120C_IMPORT_REACTIVE_ENERGY  0x004C                                      //VARh
#define SDM120C_EXPORT_REACTIVE_ENERGY  0x004E                                      //VARh
#define SDM120C_TOTAL_ACTIVE_ENERGY     0x0156                                      //Wh
#define SDM120C_TOTAL_REACTIVE_ENERGY   0x0158                                      //VARh

#define SDM630_VOLTAGE1					0x0000										//V
#define SDM630_VOLTAGE2					0x0002										//V
#define SDM630_VOLTAGE3					0x0004										//V
#define SDM630_CURRENT1					0x0006										//A
#define SDM630_CURRENT2					0x0008										//A
#define SDM630_CURRENT3					0x000A										//A
#define SDM630_POWER1					0x000C										//W
#define SDM630_POWER2					0x000E										//W
#define SDM630_POWER3					0x0010										//W
#define SDM630_POWERTOTAL				0x0034										//W`

This way i can keep both sdm120C and sdm630 registers.
Just change the baud rate and go :)
I should do some more testing when i have all 3 phases connected but it seems to work fine for the 3rd phase so i believe all will be good!

Thx a lot

reaper7, 2016-10-27:
so, when You are done all sdm630 registers, please create pull request and we add Your work.

beireken, 2016-10-27:
I have started editing your work to add the sdm630 registers.
However github is showing a lot of indent misalignments.  Is this because we use different editors?
I have tried atom and notepad++. Bot give different results. What editor are you using?

reaper7, 2016-10-28:
I have notepad++ too but my fav is a notepad2-mod:
https://github.com/XhmikosR/notepad2-mod

closed 2016-10-31 by beireken