reaper7/SDM_Energy_Meter #4: Sketch halts completely when shutting down PC

beireken, 2016-10-31:
This issue has me baffled for the passed days.
I'm using the sketch below to read from a sdm630 (or 120C with lesser registers read).

With this sketch as soon i sleep of shutdown my main desktop the esp8266 stops reading from the meter and just halts.
I have NO idea whatsoever would be the relation to my desktop pc.
The esp (wemos D1 mini) is powered by a raspberry official supply. I can disconnect the ethernet of my desktop and still the wemos keeps running fine. But as soon as i sleep it or shutdown the problem happens.

If i insert the power supply in a different outlet in the house it's still the same problem.
Any idea why this is happening? I have a different wemos reading from a different device type, which runs fine!

The problem is as soon as i shutdown my desktop i can not read the serial monitor any more, so i'm not sure how to debug this....

Any help is highly welcome! :)

`#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <SDM220_t.h>  
 
#define ASCII_ESC 27

char bufout[10];

SDM220<2400, 12, 13> sdm;

double Voltage;
double Voltage2;
double Voltage3;

double Current;
double Current2;
double Current3;

double Power;
double Power2;
double Power3;
double Power_Total;

double Frequency;

const char* ssid = "Ellast-B/G/N";
const char* password = "&gZ&c8\\J$Yu\\BSte9CFzI%Lca1Cd$9KxoutG#natBJoMmW#vj7j$1\\Tb\\g4xoOj";
const char* mqtt_server = "192.168.5.120";
const char* clientID = "NodeMCUDevKit";

char* outTopic = "BE_ESP_SDM630_Hello";

char* topic = "BE_ESP_SDM630";

const char* inTopic = "NodeMCUin";

WiFiClient espClient;
PubSubClient client(espClient);
char msg[50];

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect(clientID)) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish(outTopic, clientID);
      // ... and resubscribe
      client.subscribe(inTopic);
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void setup() {
  Serial.begin(115200);
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  sdm.begin();
}

void loop() {

  if (!client.connected()) {
    reconnect();
  }
  client.loop();
  
  sprintf(bufout,"%c[1;0H",ASCII_ESC);
  Serial.println(bufout);
  Serial.println();

  
  Voltage = sdm.readVal(SDM630_VOLTAGE1);
  delay(50);
  
  Voltage2 = sdm.readVal(SDM630_VOLTAGE2);
  delay(50);
  
  Voltage3 = sdm.readVal(SDM630_VOLTAGE3);
  delay(50);
  
  Current = sdm.readVal(SDM630_CURRENT1);
  delay(50);

  Current2 = sdm.readVal(SDM630_CURRENT2);
  delay(50);

  Current3 = sdm.readVal(SDM630_CURRENT3);
  delay(50);
  
  Power = sdm.readVal(SDM630_POWER1);
  delay(50); 

  Power2 = sdm.readVal(SDM630_POWER2);
  delay(50); 

  Power3 = sdm.readVal(SDM630_POWER3);
  delay(50);

  Power_Total = sdm.readVal(SDM630_POWERTOTAL);
  delay(50); 
  
  Frequency = sdm.readVal(SDM120C_FREQUENCY);

  
  String Publishstring = "{\"Voltage1\":" + String(Voltage, 1) + ",\"Voltage2\":"+ String(Voltage2, 1) + ",\"Voltage3\":"+ String(Voltage3, 1) + ",\"Current1\":"+ String(Current, 1) + ",\"Current2\":"+ String(Current2, 1) + ",\"Current3\":"+ String(Current3, 1)+ ",\"Power1\":"+ String(Power, 1) + ",\"Power2\":"+ String(Power2, 1) + ",\"Power3\":"+ String(Power3, 1) + ",\"Power_Total\":"+ String(Power_Total, 1) + "}";
  delay(50); 
  client.publish(topic, Publishstring.c_str(), true);
  Serial.println(Publishstring.c_str());
  Serial.println();

  
  //Serial.print(Voltage);
  //Serial.println(" V");
  //Serial.print(Voltage2);
  //Serial.println(" V");
  //Serial.print(Voltage3);
  //Serial.println(" V");
  //Serial.println();
  //Serial.print(Current);
  //Serial.println(" A");
  //Serial.print(Current2);
  //Serial.println(" A");
  //Serial.print(Current3);
  //Serial.println(" A");
  //Serial.println();
  //Serial.print(Power);
  //Serial.println(" W");
  //Serial.print(Power2);
  //Serial.println(" W");
  //Serial.print(Power3);
  //Serial.println(" W");
  //Serial.println();
  //Serial.print(Power_Total);
  //Serial.println(" W");
  //Serial.println();
  //Serial.print(Frequency);
  //Serial.println(" Hz");
  //Serial.println();
  //Serial.println();

  delay(5000);

 
  }
`

reaper7, 2016-10-31:
strange...
I have esp connected only to sdm without any uart and it works without any problem from 08-08-2016
(send values to thingspeak every minute and provides simple webserver with json)
Powered from UPS & 12V power supply via old POE line terminated by popular 12V to 5V stepdown module.

about code block - not necessary, this only print (to uart) command ESCAPE, which in true terminal programs move cursor to "HOME" (left top) position.

beireken, 2016-11-05:
I don't think it's a power issue. I tried powering from a power bank. Same issue.
I think i'll indeed buy a oled shield to debug...

In the example you do not have a delay between reading registers.
If i do this without delay(50), i get false readings. It work with you without delay?
Maybe because the sdm120 has lower baud rate default?

Also, just to try and understand, why are you using softwareserial?
Cant we use the hardware serial line and use "normal" serial library?

reaper7, 2016-11-05:
I do not think about power issue but about some peaks on uart - You have issue when wemos is connected to pc via uart line?

Example works for me as is without any problem.

Why software serial? because I need hardware serial for other things...
but You can try to change software serial to hardware in Your local copy of repo.

beireken, 2016-11-05:
Ok.. i think i got this sorted..
I got a usb to uart lying around and used this to connect to the wemos while my PC was turned off.

Seems all along it was a wifi problem... Still don't know why though...
I connected to a different ssid and now it works great!

No need to change to hardware serial. Was just out of interest!
Thx for the help.

closed 2016-11-05 by beireken