reaper7/SDM_Energy_Meter #6: direct reading of RS485 without converter

ilgrank, 2017-01-06:
Hi
At low bitrates (up to 9600baud) shouldn't the ESP8266 alone be enough to read an RS485 without a converter?
https://github.com/plieningerweb/esp8266-software-uart
https://github.com/peninquen/Modbus-Energy-Monitor-Arduino

thanks for your work!

reaper7, 2017-01-06:
No, RS485 uses two wire differential transmission.

Project:
https://github.com/plieningerweb/esp8266-software-uart
for rs485 transmission, still must use a MAX485 converter:
https://github.com/plieningerweb/esp8266-software-uart#how-to-use-rs485-support

Modbus-Energy-Monitor-Arduino also uses converter:
https://github.com/peninquen/Modbus-Energy-Monitor-Arduino/blob/master/dev%20version/esp8266modbusEnergyMonitor/esp8266-sdm120c-Aledav.jpg

ilgrank, 2017-01-06:
Thanks for the quick and kind update, I had misunderstood that
Again, many thanks!

closed 2017-01-06 by ilgrank