reaper7/SDM_Energy_Meter #8: Can you please explain unstable reading?

agsochi, 2017-04-09:
I have [this converter](http://www.ebay.com/itm/TTL-485-automatic-flow-control-module-UART-Level-Converter-3-3-5v-M92-/301724563376?hash=item46402f77b0) and Wemos D1 mini.
And i got NaN in output always in different places.
What i should tune in fw?
```
----------------------------------------
Voltage:   220.50 V
Current:   nan A
Power:     5.40 W
Total active energy: nan Wh
Total reactive energy: 0.28 VARh
----------------------------------------
Voltage:   222.70 V
Current:   nan A
Power:     5.40 W
Total active energy: nan Wh
Total reactive energy: 0.28 VARh
----------------------------------------
Voltage:   221.90 V
Current:   nan A
Power:     5.40 W
Total active energy: nan Wh
Total reactive energy: 0.28 VARh
----------------------------------------
Voltage:   222.30 V
Current:   0.03 A
Power:     nan W
Total active energy: 0.62 Wh
Total reactive energy: nan VARh
----------------------------------------
Voltage:   222.90 V
Current:   nan A
Power:     5.40 W
Total active energy: nan Wh
Total reactive energy: 0.28 VARh
----------------------------------------
Voltage:   224.50 V
Current:   nan A
Power:     5.50 W
Total active energy: nan Wh
Total reactive energy: 0.28 VARh
----------------------------------------
Voltage:   224.30 V
Current:   0.03 A
Power:     nan W
Total active energy: 0.62 Wh
Total reactive energy: nan VARh
----------------------------------------

```

beireken, 2017-04-09:
Can you try adding a delay between readings?
Like explained in this issue:
https://github.com/reaper7/SDM_Energy_Meter/issues/7

agsochi, 2017-04-09:
Yes, 50ms delay fix this. Thank you!

closed 2017-04-09 by agsochi

reaper7, 2017-05-02:
someone can test this example (without any delays between readings) and report whether problem with NaN readings still persists

```
#include <SDM.h>                                                                //import SDM template library

//SDM<2400, 12, 13> sdm;                                                        //SDM120T	baud, rx pin, tx pin
//SDM<4800, 12, 13> sdm;                                                        //SDM220T	baud, rx pin, tx pin
//SDM<9600, 12, 13> sdm;                                                        //SDM630	baud, rx pin, tx pin
//or without parameters (default from SDM.h will be used): 
SDM<> sdm;

void setup() {
  Serial.begin(115200);                                                         //initialize serial
  sdm.begin();                                                                  //initalize SDM220 communication baudrate
}

void loop() {
  float v = sdm.readVal(SDM220T_VOLTAGE);                                       //read voltage
  float c = sdm.readVal(SDM220T_CURRENT);                                       //read current
  float p = sdm.readVal(SDM220T_POWER);                                         //read power
  float f = sdm.readVal(SDM220T_FREQUENCY);                                     //read frequency
  
  Serial.printf("\nVoltage:   %.2fV\nCurrent:   %.2fA\nPower:     %.2fW\nFrequency: %.2fHz\n", v, c, p, f);

  delay(1000);                                                                  //wait a while before next loop
}
```

sjfaustino, 2017-05-07:
Using latest version of this library, latest software serial and the code below (printf on the esp8266 doesn't understand floats), I get:

Voltage:   228.20V
Current:   nanA
Power:     0.00W
Frequency: nanHz

Voltage:   228.40V
Current:   nanA
Power:     0.00W
Frequency: nanHz

Voltage:   228.40V
Current:   nanA
Power:     0.00W
Frequency: nanHz

Voltage:   228.10V
Current:   nanA
Power:     0.00W
Frequency: nanHz

````C
#include <SDM.h>                                                                //import SDM template library

SDM<2400, 12, 13> sdm;                                                        //SDM120T  baud, rx pin, tx pin
//SDM<4800, 12, 13> sdm;                                                        //SDM220T baud, rx pin, tx pin
//SDM<9600, 12, 13> sdm;                                                        //SDM630  baud, rx pin, tx pin
//or without parameters (default from SDM.h will be used): 
//SDM<> sdm;

void setup() {
  Serial.begin(115200);                                                         //initialize serial
  sdm.begin();                                                                  //initalize SDM220 communication baudrate
}

void loop() {
  float v = sdm.readVal(SDM120C_VOLTAGE);                                       //read voltage
  float c = sdm.readVal(SDM120C_CURRENT);                                       //read current
  float p = sdm.readVal(SDM120C_POWER);                                         //read power
  float f = sdm.readVal(SDM120C_FREQUENCY);                                     //read frequency
  
  Serial.printf("\nVoltage:   %sV\nCurrent:   %sA\nPower:     %sW\nFrequency: %sHz\n", String(v, 2).c_str(), String(c, 2).c_str(), String(p, 2).c_str(), String(f, 2).c_str());

  delay(1000);                                                                  //wait a while before next loop
}
`
If I modify the library with a delay(1), it works

      delay(1);
      if (_dere_pin != NOT_A_PIN)                                               //transmit to SDM  -> DE Enable, /RE Disable (for control MAX485)
        digitalWrite(_dere_pin, HIGH);

Voltage:   231.40V
Current:   0.00A
Power:     0.00W
Frequency: 50.00Hz

Voltage:   230.90V
Current:   0.00A
Power:     0.00W
Frequency: 50.04Hz

Voltage:   231.10V
Current:   0.00A
Power:     0.00W
Frequency: 50.00Hz

Voltage:   231.00V
Current:   0.00A
Power:     0.00W
Frequency: 50.00Hz

reaper7, 2017-05-07:
done

btw, which version does not understand float??
I'm using printf with %f without problem

codmpm, 2018-04-07:
I'm still getting `nan` with an SDM220, using ESP8266 core 2.4.1 and the code from https://github.com/reaper7/SDM_Energy_Meter/issues/8#issuecomment-298623158 together with the current master of the library.

```
void loop() {
  float v = sdm.readVal(SDM220T_VOLTAGE);                                       //read voltage
  float c = sdm.readVal(SDM220T_CURRENT);                                       //read current
  float p = sdm.readVal(SDM220T_POWER);                                         //read power
  float f = sdm.readVal(SDM220T_FREQUENCY);                                     //read frequency
  
  Serial.printf("\nVoltage:   %sV\nCurrent:   %sA\nPower:     %sW\nFrequency: %sHz\n", String(v, 2).c_str(), String(c, 2).c_str(), String(p, 2).c_str(), String(f, 2).c_str());

  delay(1000);  
}
```
Gives me: 
```
Voltage:   225.68V
Current:   nanA
Power:     nanW
Frequency: nanHz

Voltage:   225.59V
Current:   nanA
Power:     nanW
Frequency: nanHz
```

Adding a delay of 20ms between each `readVal()` makes the situation better, but there are still `nan`'s from time to time:

```
Voltage:   225.28V
Current:   0.00A
Power:     0.00W
Frequency: 50.01Hz

Voltage:   225.32V
Current:   0.00A
Power:     nanW
Frequency: 50.01Hz

Voltage:   225.37V
Current:   nanA
Power:     0.00W
Frequency: 50.01Hz
```

Maybe you have an idea how to optimize this. Or should I simply use the hardware serial?

Thank you for the lib. Cheers,
Patrik

reaper7, 2018-04-07:
Please try to use hw serial, I don't have more than 1-3 errors per day for sdm220 with hw serial (without any delay between readings)

reaper7, 2018-04-08:
or maybe try to remove rs485 line terminator as @da3020 wrote: https://github.com/reaper7/SDM_Energy_Meter/issues/16

codmpm, 2018-04-08:
I'm using a plain [SN75176B]( http://www.ti.com/lit/ds/symlink/sn75176b.pdf) - which is a drop-in replcament for the MAX485 - with no additional resistors.

Maybe I should add some Pull-Up's like these chinese boards have?

[image]

reaper7, 2018-04-08:
It's hard to say for me...it's just an idea.

codmpm, 2018-04-08:
I've ordered the [SP3485EN-L](https://www.exar.com/ds/sp3485_100_061912.pdf), which is a 3.3V version of the MAX485. The MAX485/SN75176B itself is specified for 5V. 

I will test when they arive to know if it's simply a problem with 3.3V logic.

reaper7, 2018-04-09:
if You waiting for new converter, please try on "old" 5V how it works on ESP8266 HW serial lines.
Of course if You have free time :)

codmpm, 2018-04-12:
Sorry for getting back late... I've switched to the SP3485EN-L (3.3V RS485 tranceiver) and have the same result, but slightly more stable. Only every 3rd/4th reading has a `nan`, independet from if I add `delay(20)` or not.

I will try the "hardware"-serial next and will get back to you.
Meanwhile I ordered samples from Maxim of the [MAX3483](https://datasheets.maximintegrated.com/en/ds/MAX3483-MAX3491.pdf) to check if these work better...

reaper7, 2018-04-13:
I have one more idea to check,
please comment lines 
from 191 to 193
```
// #if !defined ( USE_HARDWARESERIAL )
//       sdmSer.listen();                                                          //enable softserial rx interrupt
// #endif
```
and
from 251 to 253
```
// #if !defined ( USE_HARDWARESERIAL )
//       sdmSer.end();                                                             //disable softserial rx interrupt
// #endif
```
maybe something is wrong with enabling/disabling softwareserial RX interrupt

codmpm, 2018-04-14:
I still get nan-readings... and a panic:

```Voltage:   223.70V
Current:   nanA
Power:     0.00W
Frequency: 50.00Hz

Exception (0):
epc1=0x402275a0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4022642d depc=0x00000000

ctx: sys 
sp: 3ffff6c0 end: 3fffffb0 offset: 01a0
```

reaper7, 2018-04-14:
with sw serial and commented lines?
I have no idea...

codmpm, 2018-04-14:
I have to make sure, that my esp is not faulty... I will get back to you as soon as I have time to test... especially with hardware serial. I only testet it with softserial, yet...

Thanks for your time!

reaper7, 2018-04-15:
Nice to hear.
I reflash by OTA

codmpm, 2018-04-15:
Ok, I've implemented that already... finishing up the code to push some vars to mqtt. Will post it in a minute.
Powered with 3.3V directly and I'm still using softserial: getting `nan` occasionaly, but not many.

Currently I think the `nan` readings have to do something with the power source. 
Powered with an LDO ([AP2210K](https://www.diodes.com/assets/Datasheets/AP2210.pdf)) I have many more `nan`'s. I will hook the scope up to make a better assumption.

Currently I don't have `nan`'s with the LDO and an added 330µF capacitor over the 3.3V line.

codmpm, 2018-04-15:
Here is what I came up with: https://github.com/codmpm/esp-sdm-mqtt

reaper7, 2018-04-15:
I'm using MeanWell din rail 5V power supply, connected directly to wemos d1 usb connector and converter powered from wemos 3.3V pin.
Exactly like in this picture:
https://user-images.githubusercontent.com/4134851/27516122-349b6124-59b3-11e7-8151-d3e8bafbced6.jpg

codmpm, 2018-04-15:
Hm, I can't see anything faulty with the scope. Meanwhile I'm using hardware serial, which works like a charm.

Egyras, 2018-04-16:
I'am using meanwell hdr-15-5 and connected 2 x Wemos D1 mini with https://www.aliexpress.com/item/TTL-turn-RS485-module-485-to-serial-UART-level-mutual-conversion-hardware-automatic-flow-control/32709226383.html converters. Both pairs shares same power supply.
Seems only 1 pair works pretty stable and second one doesnt want to work normaly, both RX/TX leds blinks but rate is much lower..... Yesterday for half day both worked without isssues. Any suggestions, maybe just connect seperate power supply for second pair ?  Power should be enough as meanwell hdr-15-5 can provide 2.4A

reaper7, 2018-04-16:
you can read up to 254 devices from one pair (wemos+converter).
all sdm must be set to the same baudrate but must have dofferent id's
then you read one by one with: 
sdm.readVal(SOME_REGISTER, 0x01); //for first sdm
sdm.readVal(SOME_REGISTER, 0x02); //for second sdm
etc...

Egyras, 2018-04-16:
I pass all values via mqqt and need stability, will try to read several SDM from one pair.

reaper7, 2018-04-16:
I also pass values (but from one sdm) to home mqtt server and to remote thingspeak, the same values are available via internal esp8266 http server for live page and json for remote php requests, everything is done by one wemos module

commit b314e4cc (2017-04-10): Fix #8

https://github.com/reaper7/SDM_Energy_Meter/issues/8