reaper7/SDM_Energy_Meter #10: Multi meters sdm120c

mariusz84pl, 2017-06-17:
Is sdm120.h library configured to read more then 1 meter? 
Cant find in library anything about this.
I have 4 sdm120c meters and i want to read values from all of them but it reads only from meter with id1.

reaper7, 2017-06-17:
this lib is configured to read only one meter:
https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.h#L32

I can't test for more slave devices because I have only one :)

...but I commit little mod for test:
try to use **readVal** with two parameters, first -> register to read, and second(addotional) -> node Id
for eg.:
```
readVal(SDM220T_VOLTAGE, 0x01) //read voltage from node 1
readVal(SDM220T_VOLTAGE, 0x02) //read voltage from node 2
```

remember, all devices must be set with the same parameters like baudrate, parity etc except Id, which must be different for each device.

mariusz84pl, 2017-06-18:
Had to make some changes in SDM.h library because this 

`           
readVal(SDM220T_VOLTAGE, 0x01) //read voltage from node 1
      `

didnt want to work. Right now i have 4 meters and testing it but im geting few errors with value nan.
In 4464 lines i have 15 "nan" values.
Any advice what can i do to avoid them and get all reads correct? 

In this lib https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.h

added/changed:
https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.h#L163

` 
float readVal(uint16_t reg, uint8_t node) `{`
 ` 

so i can now in sketch write 

` 
sdm.readVal(SDM220T_REACTIVE_APPARENT_POWER, 0x02)
 `

from 0x01 to 0x04

reaper7, 2017-06-18:
please download full SDM.h file because changes are done in 3 places (not only in line 163 but 166 and 221 too).
For test I change my sdm dev id to 002 and it's works for me

mariusz84pl, 2017-06-18:
When i use this full SDM.h without any changes im geting this errors while it compile

`
In file included from C:\Users\Mariusz\AppData\Local\Temp\arduino_modified_sketch_760893\sketch_jun16a.ino:23:0:

C:\Users\Mariusz\Documents\Arduino\libraries\sdm-reaper/SDM2.h: In member function 'float SDM<_speed, _rx_pin, _tx_pin, _dere_pin>::readVal(uint16_t, uint8_t)':

C:\Users\Mariusz\Documents\Arduino\libraries\sdm-reaper/SDM2.h:185:14: error: 'class SoftwareSerial' has no member named 'listen'

       sdmSer.listen();                                                          //enable softserial rx interrupt

              ^

C:\Users\Mariusz\Documents\Arduino\libraries\sdm-reaper/SDM2.h:244:14: error: 'class SoftwareSerial' has no member named 'end'

       sdmSer.end();                                                             //disable softserial rx interrupt

              ^

exit status 1
Błąd kompilacji dla płytki NodeMCU 1.0 (ESP-12E Module).`

Should i use some outher SoftwareSerial lib then standard one?

when i comment in SDM.h this:

`#if !defined ( USE_HARDWARESERIAL )
      sdmSer.listen();                                                          //enable softserial rx interrupt
#endif`

and this:

`#if !defined ( USE_HARDWARESERIAL )
      sdmSer.end();                                                             //disable softserial rx interrupt
#endif`

then i can compile and upload sketch

And testing now how many "nan" values will i get.

reaper7, 2017-06-18:
I use this one swserial:
https://github.com/plerup/espsoftwareserial

mariusz84pl, 2017-06-18:
Ok changed this lib and uploaded sketch without errors so will see in 1 houer if there will by any nan values.

reaper7, 2017-06-18:
with swserial probably yes, I use hardwareserial

mariusz84pl, 2017-06-18:
For 6768 lines in file with readings have 21 nan values. :/ a litlebit to much 
Any ideas besides rising delay in SDM.h?

reaper7, 2017-06-18:
My suggestion for You is switch to hardware serial instead softwareserial

reaper7, 2017-06-18:
I do not have any errors

reaper7, 2017-06-18:
Question about this "Issue" is: this lib works for You config with multiple meters?
If Yes please comment and close this "issue"

mariusz84pl, 2017-06-18:
Yes this lib works now ok. Now only have to find out how to get rid of this errors.

closed 2017-06-18 by mariusz84pl

reaper7, 2018-06-12:
if you're talking about an **sdm_simple** example
then simple remove or increase last value (in this case 2 decimal places):
Serial.print(sdm.readVal(SDM220T_VOLTAGE), **2**);

I do not think that the library itself limit the number of decimal places.
SDM meters send reply as IEEE-754 float (32bits = 4bytes) 
and the library code also returns it without any changes or rounding