reaper7/SDM_Energy_Meter #11: Expected polling performance

apreb, 2017-11-04:
Hi, thank you for sharing this repository.

Can this meter reply to  polling rates of 1 second or below? Is there any way to get all the data at once?

Bets regards

reaper7, 2017-11-05:
Currently one register at a time(request).

But You can read all selected registers in loop (for, while...):
```
const uint8_t MYREGCNT = 5;
uint16_t myRegisters[MYREGCNT] = {SDM220T_VOLTAGE, SDM220T_CURRENT, SDM220T_POWER, SDM220T_POWER_FACTOR, SDM220T_FREQUENCY};
float myResults[MYREGCNT] = {0, 0, 0, 0, 0};

void readAllMyRegisters() {
  for (uint8_t i = 0; i < MYREGCNT; i++) {
    myResults[i] = sdm.readVal(myRegisters[i]);
    yield();
  }
}
```

apreb, 2017-11-05:
Thank you for you prompt reply.
From your experience can the meter keep up with 1-second polling of { VOLTAGE; CURRENT; POWER } ?

best regards

reaper7, 2017-11-05:
reading all 14 sdm220 registers + uploading data to thingspeak -> below 500ms (483-490)

apreb, 2017-11-05:
That's great. Thanks!

closed 2017-11-05 by apreb