reaper7/SDM_Energy_Meter #12: problem compiling ide arduino 1.8.5 for esp8266

crywolf87, 2017-11-08:
hello,
I try complile the sdm_live.page.ino with arduino ide 1.8.5 version.

But the program response always this problem.
I search the solution but i don't  have

`Arduino:1.8.5 (Windows 7), Scheda:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':

C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]

     tcp_recv(_pcb, &_s_recv);

                            ^

In file included from C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\11617752\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]

 void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;

                  ^

C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:78:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, uint16_t) {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]

     tcp_sent(_pcb, &_s_sent);

                            ^

In file included from C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\11617752\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]

 void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;

                  ^

C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:79:28: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]

     tcp_err(_pcb, &_s_error);

                            ^

In file included from C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\11617752\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]

 void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;

                  ^

C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:80:31: error: invalid conversion from 'long int (*)(void*, tcp_pcb*)' to 'tcp_poll_fn {aka signed char (*)(void*, tcp_pcb*)}' [-fpermissive]

     tcp_poll(_pcb, &_s_poll, 1);

                               ^

In file included from C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\11617752\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]

 void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;

                  ^

C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':

C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:128:25: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]

[... 115 lines cut ...]

C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:839:30: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, long int)' to 'tcp_accept_fn {aka signed char (*)(void*, tcp_pcb*, signed char)}' [-fpermissive]

   tcp_accept(_pcb, &_s_accept);

                              ^

In file included from C:\Users\11617752\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\11617752\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:317:18: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]

 void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;

                  ^

exit status 1
`

Could you Please help me?

thanks in advantage.

p.s. sorry for my easy English.

Davide

reaper7, 2017-11-08:
arduino ide version is less important but which version of ESPAsyncTCP You are using?
I've checked on last github master branch version of ESPAsyncTCP (1.1.0) and current master branch of arduino ESP8266
For me works ok, please check Your ESPAsyncTCP library.

```
C:\PROGRAMY\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\PROGRAMY\Arduino\hardware -hardware C:\Users\reape\AppData\Local\Arduino15\packages -tools C:\PROGRAMY\Arduino\tools-builder -tools C:\PROGRAMY\Arduino\hardware\tools\avr -tools C:\Users\reape\AppData\Local\Arduino15\packages -built-in-libraries C:\PROGRAMY\Arduino\libraries -libraries C:\PROGRAMY\arduino_projekty\libraries -fqbn=esp8266com:esp8266:nodemcuv2:CpuFrequency=80,FlashSize=4M3M,LwIPVariant=open,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path C:\tmp\arduino_build_954860 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\PROGRAMY\arduino_projekty\libraries\SDM_Energy_Meter\examples\sdm_live_page\sdm_live_page.ino
C:\PROGRAMY\Arduino\arduino-builder -compile -logger=machine -hardware C:\PROGRAMY\Arduino\hardware -hardware C:\Users\reape\AppData\Local\Arduino15\packages -tools C:\PROGRAMY\Arduino\tools-builder -tools C:\PROGRAMY\Arduino\hardware\tools\avr -tools C:\Users\reape\AppData\Local\Arduino15\packages -built-in-libraries C:\PROGRAMY\Arduino\libraries -libraries C:\PROGRAMY\arduino_projekty\libraries -fqbn=esp8266com:esp8266:nodemcuv2:CpuFrequency=80,FlashSize=4M3M,LwIPVariant=open,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path C:\tmp\arduino_build_954860 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\PROGRAMY\arduino_projekty\libraries\SDM_Energy_Meter\examples\sdm_live_page\sdm_live_page.ino
Using board 'nodemcuv2' from platform in folder: C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266
Using core 'esp8266' from platform in folder: C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266
Detecting libraries used...
Generating function prototypes...
Kompilowanie szkicu...
Compiling libraries...
Compiling library "ESP8266WiFi"
Compiling library "ESP8266mDNS"
Compiling library "ArduinoOTA"
Compiling library "ESPAsyncTCP"
Compiling library "ESPAsyncWebServer"
Compiling library "Time"
Compiling library "NtpClient"
Compiling library "SDM_Energy_Meter"
Compiling library "espsoftwareserial"
Compiling library "Hash"
Compiling core...
Linking everything together...
Użycie biblioteki ESP8266WiFi w wersji 1.0 z folderu: C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi 
Użycie biblioteki ESP8266mDNS z folderu C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS (legacy)
Użycie biblioteki ArduinoOTA w wersji 1.0 z folderu: C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\ArduinoOTA 
Użycie biblioteki ESPAsyncTCP w wersji 1.1.0 z folderu: C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncTCP 
Użycie biblioteki ESPAsyncWebServer w wersji 1.1.0 z folderu: C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncWebServer 
Użycie biblioteki Time w wersji 1.5 z folderu: C:\PROGRAMY\arduino_projekty\libraries\Time 
Użycie biblioteki NtpClient w wersji 2.0.5 z folderu: C:\PROGRAMY\arduino_projekty\libraries\NtpClient 
Użycie biblioteki SDM_Energy_Meter w wersji 1.0.3 z folderu: C:\PROGRAMY\arduino_projekty\libraries\SDM_Energy_Meter 
Użycie biblioteki espsoftwareserial w wersji 1.0 z folderu: C:\PROGRAMY\arduino_projekty\libraries\espsoftwareserial 
Użycie biblioteki Hash w wersji 1.0 z folderu: C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\Hash 
Szkic używa 304971 bajtów (29%) pamięci programu. Maksimum to 1044464 bajtów.
Zmienne globalne używają 36976 bajtów (45%) pamięci dynamicznej, pozostawiając 44944 bajtów dla zmiennych lokalnych. Maksimum to 81920 bajtów.
```

crywolf87, 2017-11-08:
I have verify version of version of ESPAsyncTCP.
My version in correctly 1.1.0.
Arduino Ide version is 1.6.9.

reaper7, 2017-11-08:
try to compile example from this lib:
ESPAsyncTCP -> SyncClient
and one from ESPAsyncWebServer -> ESP_AsyncFSBrowser

They compile correctly without errors?

crywolf87, 2017-11-08:
the response is error:

Arduino:1.6.9 (Windows 7), Scheda:"WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files\Arduino\hardware" -hardware "C:\Users\davide\AppData\Local\Arduino15\packages" -tools "C:\Program Files\Arduino\tools-builder" -tools "C:\Program Files\Arduino\hardware\tools\avr" -tools "C:\Users\davide\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files\Arduino\libraries" -libraries "C:\Users\davide\Documents\Arduino\libraries" -fqbn=esp8266:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=921600,FlashSize=4M3M -ide-version=10609 -build-path "C:\Users\davide\AppData\Local\Temp\buildfb990420a11fb51356058c22a3c0993b.tmp" -warnings=default -prefs=build.warn_data_percentage=75 -verbose "C:\Users\davide\Documents\Arduino\libraries\ESPAsyncWebServer-master\examples\ESP_AsyncFSBrowser\ESP_AsyncFSBrowser.ino"
C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files\Arduino\hardware" -hardware "C:\Users\davide\AppData\Local\Arduino15\packages" -tools "C:\Program Files\Arduino\tools-builder" -tools "C:\Program Files\Arduino\hardware\tools\avr" -tools "C:\Users\davide\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files\Arduino\libraries" -libraries "C:\Users\davide\Documents\Arduino\libraries" -fqbn=esp8266:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=921600,FlashSize=4M3M -ide-version=10609 -build-path "C:\Users\davide\AppData\Local\Temp\buildfb990420a11fb51356058c22a3c0993b.tmp" -warnings=default -prefs=build.warn_data_percentage=75 -verbose "C:\Users\davide\Documents\Arduino\libraries\ESPAsyncWebServer-master\examples\ESP_AsyncFSBrowser\ESP_AsyncFSBrowser.ino"
"C:\Users\davide\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\davide\AppData\Local\Temp\buildfb990420a11fb51356058c22a3c0993b.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10609 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\d1_mini" "C:\Users\davide\AppData\Local\Temp\buildfb990420a11fb51356058c22a3c0993b.tmp\sketch\ESP_AsyncFSBrowser.ino.cpp" -o "nul"
"C:\Users\davide\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\davide\AppData\Local\Temp\buildfb990420a11fb51356058c22a3c0993b.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10609 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\d1_mini" "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" "C:\Users\davide\AppData\Local\Temp\buildfb990420a11fb51356058c22a3c0993b.tmp\sketch\ESP_AsyncFSBrowser.ino.cpp" -o "nul"
"C:\Users\davide\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\davide\AppData\Local\Temp\buildfb990420a11fb51356058c22a3c0993b.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10609 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\d1_mini" "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS" "C:\Users\davide\AppData\Local\Temp\buildfb990420a11fb51356058c22a3c0993b.tmp\sketch\ESP_AsyncFSBrowser.ino.cpp" -o "nul"
[... cut ...] ino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ArduinoOTA" "-IC:\Users\davide\Documents\Arduino\libraries\ESPAsyncTCP-master\src" "-IC:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" "C:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp" -o "C:\Users\davide\AppData\Local\Temp\build0f28ab5e8de7c532d00eaa3191a83824.tmp\libraries\ESP8266WiFi\ESP8266WiFiSTA.cpp.o"

reaper7, 2017-11-08:
as You see this is not problem with SDM_Energy_Meter lib but ESPAsyncTCP or/and ESPAsyncWebServer!
Make sure that You have latest!
ESPAsyncWebServer - latest update 2 days ago
ESPAsyncTCP - latest update 9 hours ago

and of course update esp8266 core version

crywolf87, 2017-11-08:
I upgrade ESPAsyncTCP and WebServer 10 minute ago.
I think miss the update esp8266 core version.
Where I find the core version ?

reaper7, 2017-11-08:
I use from github,
here a how-to:
https://github.com/esp8266/Arduino/#using-git-version

crywolf87, 2017-11-08:
I have upgrade core 8266 through this page http://esp8266.github.io/Arduino/versions/2.3.0/doc/installing.html#using-git-version.
the version install is 2.3.0 ?

now recompiling the program but the error is the same.

C:\Users\davide\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':

C:\Users\davide\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:78:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]

     tcp_recv(_pcb, &_s_recv);

                            ^

In file included from C:\Users\davide\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\davide\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]

 void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;

                  ^

C:\Users\davide\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:79:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, uint16_t) {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]

     tcp_sent(_pcb, &_s_sent);

.... ecc

I'm no expert in this field but I followed the guides found on the internet.
I would not know what else to try

reaper7, 2017-11-08:
You still on ESP8266 core 2.3.0
Look inside Arduino->Tools->Boards
maybe You have two group for ESP8266 Modules?
one for Your old 2.3.0 and new one from Your last update from github?
Of course if you did everything correctly.

crywolf87, 2017-11-08:
[image]

reaper7, 2017-11-08:
no, go to "Strumenti" and check if you have some additional "ESP Module" group

crywolf87, 2017-11-08:
[image]

sorry I have drag the error img

reaper7, 2017-11-08:
yes, You see grey color "ESP8266 Modules",
scroll this list down and check below if you have another "ESP Modules"

crywolf87, 2017-11-08:
is the one

reaper7, 2017-11-08:
in this case something is wrong with Your process of instalation github core vesrsion,
I don't know how help You

crywolf87, 2017-11-08:
I think so better to unistall total arduino and core version 2.3.0

reaper7, 2017-11-08:
best way :)
then reinstall arduino ide, chceck if You DO NOT HAVE any ESP Modules
and install github version

crywolf87, 2017-11-08:
[image]

paste github link

[image]
[image]

find esp8266 and install

reaper7, 2017-11-08:
no, no, no
you again select stable 2.3.0, very old!

Read from this section:
https://github.com/esp8266/Arduino/#using-git-version

crywolf87, 2017-11-08:
I don't understan.
where is the last version.

reaper7, 2017-11-08:
if You don't know how to use git then try to paste this link inside preference:
https://github.com/esp8266/Arduino/releases/download/2.4.0-rc2/package_esp8266com_index.json

then go back to board manager and You see that You can select newer versions 2.4.0. rc1 and latest rc2

crywolf87, 2017-11-08:
I try to compile example from this lib:
ESPAsyncTCP -> SyncClient.

the compile is Positive 
:-)

thank you very much

reaper7, 2017-11-08:
check if sdm live page compile

crywolf87, 2017-11-08:
[image]

sdm_live compile positive

:-)

reaper7, 2017-11-08:
👍 
please close this issue

crywolf87, 2017-11-08:
ok

closed 2017-11-08 by crywolf87