reaper7/SDM_Energy_Meter #14: fix readingerrcode

apreb, 2018-02-28:
Hi, i found a small bug in the error code handling while troubleshooting NAN values.

could you please fix?

In this section:

      if (stepcnt > 0) {                                                        //if error then copy temp error value to global val and increment global error counter
        //readingerrcode = stepcnt;   // comment: after first error never returns to zero
        readingerrcount++; 
      }
      readingerrcode = stepcnt;   // FIX

reaper7, 2018-03-01:
Hi,
this is not a bug but intended action.
I need to know about the last error code, even after a few correct readings, after a days :)
With the code as above, this information disappears after the next correct reading.
If You need clear _readingerrcode_ then I can add a function like _clearErrCode()_

apreb, 2018-03-01:
I understand your intention, however the overall output is not very clear, if you perform a read request you never know if the readingerrcode is related to the actual request or if it is a previous error that happnened in the past.

With the small hack i can clearly see what's happening in my debug:

(reading,errorcode)

> (240.66,0)
> (0.78,0)
> (134.24,0)
> {"Ts":37094,"M":29192,"Va":240.7,"Aa":0.78,"Pa":134,"Err":832}
> (240.67,0)
> (0.78,0)
> (0.00,1)         //  --> got a crc here while reading power
> {"Ts":37095,"M":29192,"Va":240.7,"Aa":0.78,"Pa":134,"Err":833}
> 

Anyway after knowing the intention for me it's OK either way.

reaper7, 2018-03-01:
that's why I suggest adding clearErrCode() function, then:
1. reading
2. checking getErrCode() and if <> 0 then clearErrCode()

at the moment you are reading getErrCode() anyway, so additional clearErrCode() probably not be a problem :)
and everyone will be happy

apreb, 2018-03-01:
Sure, thank you!

reaper7, 2018-03-01:
done

You can use additional clearErrCode()
or simply getErrCode(**true**) which will delete automatically

apreb, 2018-03-01:
Perfect! Thanks for your availability!! you can close this issue

closed 2018-03-01 by reaper7