reaper7/SDM_Energy_Meter #17: hardwareserial

Egyras, 2018-04-12:
hi, I have tried to define in my sketch hardwareserial
#define USE_HARDWARESERIAL 
Use Wemos D1 Mini with https://user-images.githubusercontent.com/28455705/34221336-f3462a6a-e5b7-11e7-94fa-5fbb4fc57065.jpeg
Seems that it works when I connect to 3/1 pins, doesnt work with 13/15 pins. 
Basicaly with 13/15 it "hangs" SDM120 counter as it starts blinking.... Because 15 pin on esp8266 must be low during boot up ? So I assume that 13/15 pins are not usable at all, right ?
Corect pins for hardwareserial should be only 3/1 ?

reaper7, 2018-04-12:
Yes, it's true, use pins 13/15 makes impossible to start esp properly
but it only affects the boot process, if you leave pin 15 not connected at boot
and connect it only after starting then everything works fine.
Of course, it's not practical...so pins 1/3 are the best option

reaper7, 2018-04-13:
on the converter board are located two resistors R4 and R3, I think they are pullup resistors
so 4k7 on converter board gives stronger pullup than 10k pulldown on wemos gpio15 pin.

Egyras, 2018-04-15:
what converter you are using, mine seems not very stable.....?

reaper7, 2018-04-15:
the same as you, mine is very stable, 
~2days / 603597 sdm readings and only 2 errors:
[image]

please read last discussion in [this issue](https://github.com/reaper7/SDM_Energy_Meter/issues/8#issuecomment-381398551) (to the end):

conclusions, test on another esp8266 module and check your power supply

Egyras, 2018-04-17:
Seems I had some bad connection between Wemos and converter... all runs fine 24 hours :+1:

closed 2018-04-17 by Egyras