reaper7/SDM_Energy_Meter #18: Portin to Arduino?

Franchelli, 2018-05-07:
the library works very well on Wemos D1 and I ask if it is possible to modify it to make it work with Arduino UNO or NANO.

reaper7, 2018-05-08:
I never do this but please check this fork:
https://github.com/joearkay/SDM_Energy_Meter

Franchelli, 2018-05-09:
I tried to run the test file but it seems not to compile correctly, any ideas?

reaper7, 2018-05-09:
I do not have arduino,
ask the author of the modification.

reaper7, 2018-05-10:
please try my current master branch,
init support for avr is added **but only for SoftwareSerial!**

Franchelli, 2018-05-12:
I have tested your modification on an Arduino Nano but unfortunately all the output values are "Nan" ... I used the SDM configuration `<2400, 7, 8> sdm;` and I used the RS485 converter without DR DA pin with automatic flow control. With the same configuration sw / hw and wemos D1 everything works perfectly. I hope for your interest because your library is the one that currently works best and is easy to implement

reaper7, 2018-05-12:
so ... now it works fine?

Franchelli, 2018-05-14:
yes, it works perfectly just that I do not understand the reason for the inverted connections, but now working between TX and RX.

reaper7, 2018-05-14:
I don't know too, initialization swserial lib for esp8266 and for arduino are the same:
avr:
```
SoftwareSerial(uint8_t receivePin, uint8_t transmitPin, bool inverse_logic = false);
```
esp
```
SoftwareSerial(int receivePin, int transmitPin, bool inverse_logic = false, unsigned int buffSize = 64);
```
rx is first parameter, tx second

btw, if works, please close this issue.

closed 2018-05-14 by Franchelli