reaper7/SDM_Energy_Meter #20: Baud rate problem

Paride94, 2018-06-12:
Hi reaper7 thank you very Much for your big and Very useful work ! 
However i saw that the baud rate at 4800 doesn't work , while it work at 2400 , do you confirm this ? 
I tried with sdm120C and sdm120CT (with TA) .
Another question is: how Can i put the function " acos(PF)"  to show the Phase single , by Power factor in the table ? 
This sdm120 has not info about Phase in his register only PF and i would to convert it..
Thank your for any answer :)

reaper7, 2018-06-12:
I checked the baud rate 4800 only with sdm220.
Generally, the library has nothing to do with the transmission speed,
only forward information to hardware or software serial library.

Of course, you change the transmission speed during library initialization:
```
SDM<4800, your_rx_pin, your_tx_pin> sdm;
```
and at the same time in the device (sdm120) ???

about second question...I do not know anything about it, sorry

Paride94, 2018-06-12:
Hi , thank you very much for your fast answer , i changed (in sdm live
page .ino ) only the string:

SDM<4800, 13, 15, NOT_A_PIN> sdm;

with

SDM<2400, 13, 15, NOT_A_PIN> sdm;

for sdm120C and sdm120CT , and i removed the comment command ( // #define
USE_HARDWARESERIAL ) to turn on the command: #define USE_HARDWARESERIAL
for HW communication.

while i have not  changed nothing in sdm.h library and it worked :)

Np about the second question i need to try to add the function in the table
to calculate the phase angle from power factor .. :)

ldab, 2018-06-12:
acos(PF) would return radians, try:

result = acos (param) * 180.0 / PI;

reaper7, 2018-06-12:
if You are using HARDWARE SERIAL then initialization looks different.
Look at https://github.com/reaper7/SDM_Energy_Meter#initializing
If You have sdm connected to hardware serial pins (gpio3/gpio1)
then init sdm library:
```cpp
//lib init when Hardware Serial is used:
#define USE_HARDWARESERIAL
#include <SDM.h>
//      ______________________baudrate
//     |    __________________dere pin(optional for max485)
//     |   |    ______________swap hw serial pins from 3/1 to 13/15(default false)
//     |   |   |
SDM<4800, 12, false> sdm;
```
or if You use converter with automatic flow control:
```cpp
SDM<4800> sdm;
```

Paride94, 2018-06-18:
Hi reaper7 , thank you very Much , however don't worry i have already made this Fix about the hardware serial because i use the converter with automatic control flow. 
While i Need to Try tour fix about the the convertion from PF to Phase angle , thank you again !! 
However i have another problem, I tried to modify the sdm_live_page.ino to include the command to upload the parameters on my thingspeak Channel , without success.. how Can i  make uploading of the parameters in thingspeak? Are there  library or Simply fix ? Thanks , regards

reaper7, 2018-06-19:
for thingspeak look at this library:
https://github.com/mathworks/thingspeak-arduino

something like this (part of code):
```cpp
#include <ESP8266WiFi.h>
#include <ThingSpeak.h>

WiFiClient tsclient;

unsigned long myChannelNumber = 987654;
const char * myWriteAPIKey = "ABCDEFGHIJKLMNOPR";

void sendToTS() {
  char floatchr[12];
  ThingSpeak.setField(1,dtostrf(sdmarr[VOLTAGE].regvalarr,0,2,floatchr));
  ThingSpeak.setField(2,dtostrf(sdmarr[CURRENT].regvalarr,0,2,floatchr));
  ThingSpeak.setField(3,dtostrf(sdmarr[POWER].regvalarr,0,2,floatchr));

  int res = ThingSpeak.writeFields(myChannelNumber, myWriteAPIKey);

  if (res != OK_SUCCESS) {
    // Serial.println("err");
  }
}

void setup() {
  // your setup stuff
  // connect to wifi, etc...

  ThingSpeak.begin(tsclient);

  // rest of your setup stuff

}

void loop() {
  
  // your stuff to execute:
  //sendToTS();
  // every minute

}
```

Sorry but I do not have time for other things and also this is not a discussion board.
If the problem from issue topic is resolved, please close this issue.

Paride94, 2018-06-19:
sorry , you are right ! thank you for everything and regards .

closed 2018-06-19 by Paride94