reaper7/SDM_Energy_Meter #23: Problem compiling with arduino IDE 1.8.6 and hardware serial

v-a-d-e-r, 2018-07-30:
Hello all. I'm running mad now. With software serial everything works fine, with hardware serial I can't even compile and get this error:

undefined reference to `SDM::SDM(HardwareSerial&, long, int, int, bool)'
sketch/ESPEasy.ino.cpp.o: In function `Plugin_150(unsigned char, EventStruct*, String&)':
undefined reference to `SDM::SDM(HardwareSerial&, long, int, int, bool)'
undefined reference to `SDM::SDM(HardwareSerial&, long, int, int, bool)'
collect2: error: ld returned 1 exit status

Any idea what's going wrong here?

salberin, 2018-07-30:
Came here to write a bug report, as I am experiencing the exact same error....

salberin, 2018-07-31:
I'm experiencing this using a Wemos D1 Mini. Are you using the same hardware?

v-a-d-e-r, 2018-07-31:
No. I have a Lolin NodeMCU. But I think it does not depend on the hardware. More on the communication between the SDM software and the hardware serial library.

reaper7, 2018-07-31:
Uncomment
#define USE_HARDWARESERIAL
in sdm_user_config.h file

v-a-d-e-r, 2018-07-31:
Hi. Many thanks, it works! Can't believe it..... I thought when I define it in the .ino it is global? And #define USE_HARDWARESERIAL is before include <SDM.h>!?

closed 2018-08-01 by v-a-d-e-r