reaper7/SDM_Energy_Meter #24: Error with arduino 2.4.2

helianto, 2018-08-28:
It seems that this version have their own version of software serial, and if I activate hardware serial, there is one error:

in line:
SDM sdm(Serial, 4800, NOT_A_PIN, SERIAL_8N1, false)

Se encontraron múltiples librerías para "SoftwareSerial.h"
Usado: C:\Users\javier\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\SoftwareSerial
 No usado: C:\Users\javier\Documents\Arduino\libraries\espsoftwareserial-master
exit status 1
no matching function for call to 'SDM::SDM(HardwareSerial&, int, int, SerialConfig, bool)'

reaper7, 2018-08-29:
how you activate hw serial??
it may be a similar problem with https://github.com/reaper7/SDM_Energy_Meter/issues/23#issuecomment-409142990

helianto, 2018-08-29:
Yes, i had placed #define USE_HARDWARESERIAL  into ino file.... so now is solved...

closed 2018-08-29 by helianto