reaper7/SDM_Energy_Meter #25: ESP32 and serial2

helianto, 2018-08-29:
I am trying to configure the script to use with esp32 (two hardware uart) i want to use serial0 to usb, serial1 to modbus, the source file compile with default hardware configuration, but if i create serial two:

First the source: 
```
HardwareSerial Serial1(1);

#define RXD2 16
#define TXD2 17

#include <SDM.h>                                                                //import SDM library

SDM sdm(Serial1, 4800, NOT_A_PIN, SERIAL_8N1, false);                            //HARDWARE SERIAL

void setup() {
  Serial.begin(115200);                                                         //initialize serial
   Serial1.begin(4800, SERIAL_8N1, RXD2, TXD2);
  sdm.begin();                                                                  //initialize SDM communication
  

  Serial.println("Serial Txd is on pin: "+String(TX));
Serial.println("Serial Rxd is on pin: "+String(RX));
}

void loop() {
```

the error:

C:\Users\javier\AppData\Local\Temp\arduino_build_587964/arduino.ar(HardwareSerial.cpp.o):(.bss.Serial1+0x0): multiple definition of `Serial1'

sketch\sdm_simple120c.ino.cpp.o:(.bss.Serial1+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compilando para la tarjeta ESP32 Dev Module.

helianto, 2018-08-29:
IT WORKS!!! BUT WITH SMALL CHANGES... you cannot use serial1, have to use serial 2... if use serial 1... it hangs reason: https://www.youtube.com/watch?time_continue=55&v=GwShqW39jlE

reaper7, 2018-08-29:
curently I don't have any sdm but this sketch works for me (without sdm connected):
```cpp
#include <SDM.h>                                                                //import SDM library

#define RXD2 16
#define TXD2 17

SDM sdm(Serial1, 9600, NOT_A_PIN, SERIAL_8N1, false);

void setup() {
  Serial.begin(115200);                                                         //initialize serial
  Serial1.begin(4800, SERIAL_8N1, RXD2, TXD2);
  sdm.begin();                                                                  //initialize SDM communication

  Serial.println("Serial0 Txd is on pin: "+String(TX));
  Serial.println("Serial0 Rxd is on pin: "+String(RX));

  Serial.println("Serial1 Txd is on pin: "+String(TXD2));
  Serial.println("Serial1 Rxd is on pin: "+String(RXD2));

}

void loop() {
  char bufout[10];
  sprintf(bufout, "%c[1;0H", 27);
  Serial.print(bufout);

  Serial.print("Voltage:   ");
  Serial.print(sdm.readVal(SDM220T_VOLTAGE), 2);                                //display voltage
  Serial.println("V");

  delay(50);

  Serial.print("Current:   ");
  Serial.print(sdm.readVal(SDM220T_CURRENT), 2);                                //display current  
  Serial.println("A");

  delay(50);

  Serial.print("Power:     ");
  Serial.print(sdm.readVal(SDM220T_POWER), 2);                                  //display power
  Serial.println("W");

  delay(50);

  Serial.print("Frequency: ");
  Serial.print(sdm.readVal(SDM220T_FREQUENCY), 2);                              //display frequency
  Serial.println("Hz");   

  delay(1000);                                                                  //wait a while before next loop
}
```

on Serial console I got:
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
Serial0 Txd is on pin: 1
Serial0 Rxd is on pin: 3
Serial1 Txd is on pin: 17
Serial1 Rxd is on pin: 16
[1B][1;0HVoltage:   nanV
Current:   nanA
Power:     nanW
Frequency: nanHz
[1B][1;0HVoltage:   nanV
Current:   nanA
Power:     nanW
Frequency: nanHz
[1B][1;0HVoltage:   nanV
Current:   nanA
Power:     nanW
Frequency: nanHz
```

helianto, 2018-08-29:
With my esp32 (wroom32) this code:

Backtrace: 0x400d13e8:0x3ffb1f60 0x400e5c93:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d13e8  PS      : 0x00060530  A0      : 0x800e5c96  A1      : 0x3ffb1f60  
A2      : 0x3ffc2588  A3      : 0x0800001c  A4      : 0x00060023  A5      : 0x3ffb2034  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d0a15  A9      : 0x3ffb1f40  
A10     : 0x3ffb1f74  A11     : 0x3f400ef3  A12     : 0x0800001c  A13     : 0x00000010  
A14     : 0x00000011  A15     : 0x00000000  SAR     : 0x00000016  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d13e8:0x3ffb1f60 0x400e5c93:0x3ffb1fa0

helianto, 2018-08-29:
This configuration works for me: default uart2 port configuration.

```
``#include <SDM.h>                                                                //import SDM library
#include <WiFi.h>

const char* ssid     = "puticlub juanita 4";
const char* password = "xxxxxxxxxxxxxx";

SDM sdm(Serial2, 4800, NOT_A_PIN, SERIAL_8N1, false);                            //HARDWARE SERIAL

float V1 =0;
float V2 =0;
float V3 =0;
float AM1 =0;
float AM2 =0;
float AM3 =0;
float AMT =0;
float P1 =0;
float P2 =0;
float P3 =0;
float PT =0;
float VAr1 =0;
float VAr2 =0;
float VAr3 =0;
float VArT =0;
float PF1=0;
float PF2=0;
float PF3=0;
float PFT=0;
float ANGLE1=0;
float ANGLE2=0;
float ANGLE3=0;
float ANGLET=0;
float FREC=0;
float A_AVER=0;
float V_AVER=0;
float ENERGY=0;

void setup() {
  Serial.begin(115200);                                                         //initialize serial
   Serial2.begin(4800, SERIAL_8N1);
  sdm.begin();         
```

reaper7, 2018-08-29:
Strange, try to temporary remove all Serial.println from setup section

I use arduino-esp32 current github version

helianto, 2018-08-29:
same error
I am using the same version of arduino, i think that the only solution is change the port pins into arduino code...

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4000c271  PS      : 0x00060034  A0      : 0x800855e4  A1      : 0x3ffb1e70  
A2      : 0x3ffb0024  A3      : 0x3f4013f0  A4      : 0x00000014  A5      : 0x3ffc0bc0  
A6      : 0x3ffc0c08  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x3f4013f1  
A10     : 0x000000a5  A11     : 0x00000000  A12     : 0x80086077  A13     : 0x3ffc0b90  
A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4000c271:0x3ffb1e70 0x400855e1:0x3ffb1e90 0x40086d58:0x3ffb1eb0 0x40086d0e:0x3ffb1ed0 0x400d1731:0x88b60aa5

reaper7, 2018-08-30:
you tests my sketch from above with or without sdm connected?

helianto, 2018-08-30:
Both options same result...

helianto, 2018-08-30:
Another question... off topic in this topic, i have a peacefair pzem-016 (another modbus with CT energy meter), Would it be possible to expand the library to incorporate the registers of this meter?... attached documentation
[PZEM-014,016 specification.docx](https://github.com/reaper7/SDM_Energy_Meter/files/2335702/PZEM-014.016.specification.docx)

reaper7, 2018-08-30:
OT - you must make your own copy of this repo, add registers and tests this.
I can't do this without hw.

helianto, 2018-08-30:
Of course, I will do it, only, please, and when you have time, look at the instructions I have sent in case there should be a change in the library to "read" the values ​​of the records ...

helianto, 2018-08-31:
I have added these registers to sdm630 for my own use... if you want you can add them to the library:
#define SDM630_PHASE_1_LN_VOLTS_THD         0x00EA                              //%
#define SDM630_PHASE_2_LN_VOLTS_THD         0x00EC                              //%
#define SDM630_PHASE_3_LN_VOLTS_THD         0x00EE                              //%
#define SDM630_AVERAGE_VOLTS_THD            0x00F8                              //%
#define SDM630_PHASE_1_CURRENT_THD          0x00F0                              //%
#define SDM630_PHASE_2_CURRENT_THD          0x00F2                              //%
#define SDM630_PHASE_3_CURRENT_THD          0x00F4                              //%
#define SDM630_AVERAGE_CURRENT_THD          0x00FA                              //%

helianto, 2018-10-18:
After more than 1 month working flawless I think it works without trouble

closed 2018-10-18 by helianto