reaper7/SDM_Energy_Meter #28: Problem new sdm library

crywolf87, 2019-02-15:
helllo,

I 'have upgrade a new library sdm, but the esp12F doesn't works.

**I have insert to the sketch this line like the guide:**
`#include <SDM.h>
#include <SDM_Config_User.h>
#define USE_HARDWARESERIAL

SDM sdm(Serial);

voidsetup(){
sm.begin();
}`

voidloop(){
 s = String(sdm.readVal(SDM220T_VOLTAGE, 1));
      if (s != "nan") {
      s.toCharArray(charBuf, 50);
}

**in the SDM_config_user.h**
`#define USE_HARDWARESERIAL                                                      //define USE_HARDWARESERIAL

//------------------------------------------------------------------------------

#define SDM_UART_BAUD                       2400                                //define user baudrate

//------------------------------------------------------------------------------

//------------------------------------------------------------------------------
#ifdef USE_HARDWARESERIAL

  #define SDM_UART_CONFIG                   SERIAL_8N1                          //define user SDM_UART_CONFIG

/*
  #define SWAPHWSERIAL                      0                                   //define user SWAPHWSERIAL, if true(1) then swap uart pins from 3/1 to 13/15 (only ESP8266)
*/
#endif
//------------------------------------------------------------------------------

#define MAX_MILLIS_TO_WAIT                  500                                 //define user MAX_MILLIS_TO_WAIT to wait for response from SDM
`

but the esp donn't works.

can you help me ?

thank's in advantage

Davide

reaper7, 2019-02-18:
Paste whole code (complete sketch and complete SDM_config_user.h after changes),
use correct formatting: https://help.github.com/articles/creating-and-highlighting-code-blocks/

Show the wiring diagram with ESP <-> RS485 <-> SDM

I don't understand "esp12F doesn't works"??
If esp12F doesn't works then replace it with a new one.

Egyras, 2019-03-04:
Doesn't work for me with wemos and HARDWARESERIAL, got values as Nah... With old library no issues.
I see that there is no RX and TX pin defining anymore

reaper7, 2019-03-04:
hardware serial does not have pins definition, it is always GPIO3/GPIO1.

If you want to use HARDWARESERIAL mode then you must uncomment
#define USE_HARDWARESERIAL in SDM_Config_User.h (by default it is commented)
https://github.com/reaper7/SDM_Energy_Meter#configuring

Egyras, 2019-03-05:
> hardware serial does not have pins definition, it is always GPIO3/GPIO1.
> If you want to use HARDWARESERIAL mode then you must uncomment
> #define USE_HARDWARESERIAL in SDM_Config_User.h (by default it is commented)
> https://github.com/reaper7/SDM_Energy_Meter#configuring

My SDM_Config_User.h looks like:

#define USE_HARDWARESERIAL 
#define SDM_UART_BAUD                       2400   //using SDM120
#define DERE_PIN                            NOT_A_PIN 
#ifdef USE_HARDWARESERIAL
#define SDM_UART_CONFIG                   SERIAL_8N1                        
#define SWAPHWSERIAL                      0                               
#endif
#define MAX_MILLIS_TO_WAIT                  500

reaper7, 2019-03-05:
looks OK.

it should not make a difference, but try this SDM_Config_User.h:
```
#define USE_HARDWARESERIAL
#define SDM_UART_BAUD 2400 //using SDM120
// #define DERE_PIN NOT_A_PIN
#ifdef USE_HARDWARESERIAL
// #define SDM_UART_CONFIG SERIAL_8N1
// #define SWAPHWSERIAL 0
#endif
// #define MAX_MILLIS_TO_WAIT 500
```

I also use wemos with hw serial (compiled 5 days earlier) and my works.

What type of converter you use?
Show the diagram or photo of the whole device.

Egyras, 2019-03-05:
> #define USE_HARDWARESERIAL
> #define SDM_UART_BAUD 2400 //using SDM120
> // #define DERE_PIN NOT_A_PIN
> #ifdef USE_HARDWARESERIAL
> // #define SDM_UART_CONFIG SERIAL_8N1
> // #define SWAPHWSERIAL 0
> #endif
> // #define MAX_MILLIS_TO_WAIT 500

did from scratch and all works, strange :)

reaper7, 2019-03-17:
closed due to inactivity

closed 2019-03-17 by reaper7