reaper7/SDM_Energy_Meter #29: Read out further registers sdm630

ralphi, 2019-04-11:
There is a PDF with all registers (SDM 630)
[https://www.varmepumpsforum.com/Eastron_SDM630_Modbus_protocol_V1.5.pdf](https://www.varmepumpsforum.com/Eastron_SDM630_Modbus_protocol_V1.5.pdf)

I needed other editions and I simply expanded the SDM.h:

#define SDM630_IMPORT1                		0x015a                              //Wh
#define SDM630_IMPORT2                		0x015c                              //Wh
#define SDM630_IMPORT3                		0x015e                              //Wh

#define SDM630_EXPORT1                		0x0160                              //Wh
#define SDM630_EXPORT2                		0x0162                              //Wh
#define SDM630_EXPORT3                		0x0164     

#define SDM630_TOTAL_ENERGY1           		0x0166                              //Wh
#define SDM630_TOTAL_ENERGY2                0x0168                              //Wh
#define SDM630_TOTAL_ENERGY3                0x016a                              //Wh

reaper7, 2019-04-11:
the document shows that it is kWh not Wh

ralphi, 2019-04-11:
Yes - corrected :-)

reaper7, 2019-04-11:
done!

closed 2019-04-11 by reaper7