reaper7/SDM_Energy_Meter #32: All readings are "nan", except the first reading

marcelroeder, 2019-07-03:
Hello,

i am trying to connect an nodemcu board through an rs485 transceiver to the SDM630 v2 meter over softwareserial. 

The problem is that all readings show "nan". When i perform a reset of the nodemcu board, the first reading is working and showing the correct value, but after this all other readings are "nan". After an reset of the nodemcu board i can read another value. Get error code shows me error 4.

Do i need to clear some buffers after a reading was performed?

To test the hardware i tried an arduino uno with softwareserial and an arduino mega with hardwareserial, both without any problems. Also i tried another nodemcu board, but without any help.

What am i doing wrong?

`
#include <SoftwareSerial.h>                                                     //import SoftwareSerial library
#include <SDM.h>                                                                //import SDM library

SoftwareSerial swSerSDM(12, 14);                                                //config SoftwareSerial 
SDM sdm(swSerSDM, 9600, 4);                                                                //import SDM library

void setup() {
  Serial.begin(9600);                                                         //initialize serial
  sdm.begin();                                                                  //initialize SDM communication
}

void loop() {
  char bufout[10];
  sprintf(bufout, "%c[1;0H", 27);
  Serial.print(bufout);

  Serial.print("Voltage:   ");
  Serial.print(sdm.readVal(SDM630_VOLTAGE1), 2);                                //display voltage
  Serial.println("V");

  delay(50);

  Serial.print("Current:   ");
  Serial.print(sdm.readVal(SDM630_CURRENT1), 2);                                //display current
  Serial.println("A");

  delay(50);

  Serial.print("Power:     ");
  Serial.print(sdm.readVal(SDM630_POWER1), 2);                                  //display power
  Serial.println("W");

  delay(50);

  Serial.print("Frequency: ");
  Serial.print(sdm.readVal(SDM630_FREQUENCY), 2);                              //display frequency
  Serial.println("Hz");

  delay(50);

  Serial.print("Total Power: ");
  Serial.print(sdm.readVal(SDM630_IMPORT_ACTIVE_ENERGY), 2);                              //display Total Used Power
  Serial.println("kWh");

  delay(50);

  uint16_t lasterror = sdm.getErrCode(true);
  sdm.clearErrCode();

  Serial.print("Error Code: ");
  Serial.println(lasterror);
  lasterror = 0;

  delay(1000);                                                                  //wait a while before next loop
}
`

reaper7, 2019-07-04:
Have you tried this lib on arduino uno with softwareserial? And it works correctly?

marcelroeder, 2019-07-04:
Yes i tested the uno with softwareserial and it worked. Only the esp8266 makes problems with softwareserial.

reaper7, 2019-07-04:
tnx for the info.
What version of esp softwareserial You are using?

marcelroeder, 2019-07-04:
The latest version, i just downloaded the library yesterday from your link in the introduction

deepthroatme, 2019-10-13:
Hi together, I can confirm this with SW-Serial on an ESP8266 (NodeMCU) and the sdm_live_page sample: He reads only once the Voltage and then only "zero". 
With HW-Serial it works fine and also with Serial-Swap. But at swap I have to disconnect my converter due to the described problem with pullup/down resistor. Hope to find the "wrong" resistor…

reaper7, 2019-12-31:
probably fixed with this: https://github.com/reaper7/SDM_Energy_Meter/commit/589dfa00a029d0dee0e6b1612b1fd470821c5072

closed 2020-01-10 by reaper7