reaper7/SDM_Energy_Meter #33: Add dds238 support

E-NINA, 2019-08-19:
I'm not a programmer so excuse first for some noob comment, but I've trying make a small adaptation to your library to be able to read also power meters from hiking
They also use Modbus for communication. I’ve model dds238-4 W and basically do the same as Eastrons.
My intention was to contribute and perhaps had also those kind of meters to your library.
As far I’ve seen communication protocol to request data is exactly the same structure difference is that:

SDMxxx read registers are 4X inputs and on DDS238xxx uses 3X inputs
Replay data from SDMxxx came on IEEE-754 Floating Point format and DDS238xxx date came as HEX format

See a table below with a summary of those:

[image]

[image]

I’ve made the following modifications to your library to try get some data:
SDM.h
Line 50			define SDM_B_02                            **0x04**
Lines 63…77
//SDM 220 registers  **(DDS238)**
#define SDM220T_VOLTAGE                     **0x000C**                              //V
#define SDM220T_CURRENT                     **0x000D**                              //A
#define SDM220T_POWER                       **0x000E**                              //W
**//**#define SDM220T_ACTIVE_APPARENT_POWER       0x000E                              //VA  (don’t exists on DDS238)
#define SDM220T_REACTIVE_APPARENT_POWER     **0x000F**                              //VAR
#define SDM220T_POWER_FACTOR                **0x0010**                              //
**//**#define SDM220T_PHASE_ANGLE                 0x0024                              //DEGREE  (don’t exists on DDS238)
#define SDM220T_FREQUENCY                   **0x0011**                              //Hz
#define SDM220T_IMPORT_ACTIVE_ENERGY        **0x000A**                              //Wh   (data length 0x02)
#define SDM220T_EXPORT_ACTIVE_ENERGY        **0x0008**                              //Wh   (data length 0x02)
**//**#define SDM220T_IMPORT_REACTIVE_ENERGY      0x004C                              //VARh  (don’t exists on DDS238)
**//**#define SDM220T_EXPORT_REACTIVE_ENERGY      0x004E                              //VARh  (don’t exists on DDS238)
#define SDM220T_TOTAL_ACTIVE_ENERGY         **0x0000**                              //Wh   (data length 0x02)

I compiled your example and give no error on reading but always get responses as “0,00”
How can change data to be on HEX format and not results of IEEE-754 floating point converter.

A table of holding registers for that meters you can also lock on:
https://github.com/fawno/Modbus/blob/2cd4cd89b191cc84c40457d6d3cb406953a86d19/DDS238-2%20ZN-S%20Modbus.md

Thanks
ENina

reaper7, 2019-08-19:
@E-NINA - look at this fork:
https://github.com/adlerweb/BZ40i_Energy_Meter

E-NINA, 2019-08-19:
Hello,
Thanks for the quick replay. What I seen from my “non expertise eyes” so far that might be the solution is the difference on lines:

SDM.h line 183
float readVal(uint16_t reg, uint8_t node = SDM_B_01);                       //read value from register = reg and from deviceId = node

BZ40i.h line 230
float readVal(uint16_t reg, **_byte type = 0_**, uint8_t node = BZ40i_B_01);      //read value from register = reg and from deviceId = node

and 

SDM.cpp line 64
float SDM::readVal(uint16_t reg, uint8_t node) {

BZ40i.cpp line 47
float BZ40i::readVal(uint16_t reg, **_byte type_**, uint8_t node) {

This evening I will tray add this 2 arguments “byte type = 0” to SDM.h and “byte type” to SDM.cpp and also try to change your recommended fork from BZ40i.
I will let you now about the results.

Thanks
ENina

reaper7, 2019-08-19:
a much more important difference is the correct assignment of received data :)
**type** variable specifies which method to use
and methods are included from the line 111:
https://github.com/adlerweb/BZ40i_Energy_Meter/blob/master/BZ40i.cpp#L111

BTW
Personally I will not add a device unknown for me,
I suggest you clone this library and adapt it to your requirements.

E-NINA, 2019-08-19:
Hello again,

thanks for your helpful input, I don’t really understand mathematics beyond variables types but I managed:

so as there are 2 types of data lengths from different registers outputs I’ve made the following:
Keep the 2 arguments “byte type = 0” to SDM.h and “byte type” to SDM.cpp
So I can define one or the other types of output registers and for arguments I put:

if(type == 2) {
		 int32_t sinput = 0;
            ((uint8_t*)&sinput)[1]= sdmarr[3];
            ((uint8_t*)&sinput)[0]= sdmarr[4];
            res = sinput;
		}else{
            int32_t sinput = 1;
            ((uint8_t*)&sinput)[3]= sdmarr[3];
            ((uint8_t*)&sinput)[2]= sdmarr[4];
            ((uint8_t*)&sinput)[1]= sdmarr[5];
            ((uint8_t*)&sinput)[0]= sdmarr[6];
            res = sinput;
			}		

I will try on next day clone this library and make this available to others, let see I can manage.
Thanks a lot for your good hints

Best regards
ENina

E-NINA, 2019-08-20:
Hello

I think I managed, forked complete repository changed all variables and seems everything is working.
Only stupid question how I can put dds238_Energy_Meter repository visible to everyone?

Best regards
ENina

reaper7, 2019-08-21:
I don't know, I've never had to do special actions to make it visible

closed 2019-08-21 by E-NINA