reaper7/SDM_Energy_Meter #35: ESP32 with SoftwareSerial

micha6554, 2019-09-27:
Its ESP32 with SoftwareSerial possible? 

Thanks

reaper7, 2019-09-27:
Theoretically yes, why not...
but I've never tested it.

you can be first :)

micha6554, 2019-09-27:
Its from ESP32 the pin 16/17 RX2/TX2 Soft- or  Hardwareserial?

reaper7, 2019-09-27:
don't ask me, I don't know, you have to understand it yourself

reaper7, 2019-09-29:
Personally I do not recommend using swserial, especially for esp32 where we have 3 hardware uarts.
...so, for the test, if you have hwserial 2
1. uncomment in SDM_Config_User.h:
```
#define USE_HARDWARESERIAL
```
2. in sdm_simple/sdm_simple.ino example:
```
//#include <SoftwareSerial.h> // < comment out
#include <SDM.h>

//SoftwareSerial swSerSDM(13, 15); // < comment out

SDM sdm(Serial2, 9600, NOT_A_PIN, SERIAL_8N1, 16, 17); //config SDM
```
(of course, modify the baud rate if you have other than 9600)

basically if you use the default uart pins, then initialization may look simpler:
```
SDM sdm(Serial2, 9600, NOT_A_PIN)
```

micha6554, 2019-10-01:
****************************
Thanks reaper7!!!!
*****************************

Its work!!

// *************************************************************************************************************
// modbus for SDM-120  Energiemesser
// https://github.com/reaper7/SDM_Energy_Meter
// Config unter https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM_Config_User.h (unkomment #define USE_HARDWARESERIAL !!!!!!!!!)
// *************************************************************************************************************
#define USE_HARDWARESERIAL                                            // hardwareserial|softwareserial
#define BAUDRATE            2400                                      // 2400bps (SDM-120)
#define MAX3485_RE_DE_PIN   GPIO18                                    // RE-DE pin
#define RX                  GPIO16
#define TX                  GPIO17
#include <SDM.h>                                                      // SDM library
SDM sdm(Serial2, BAUDRATE, MAX3485_RE_DE_PIN, SERIAL_8N1, RX, TX);    // (Serial2, baudrate, MAX3485_RE_DE_PIN, SERIAL_8N1, RX, TX)

reaper7, 2019-10-01:
I'm glad to hear that :)
You are the first who confirmed :D

closed 2020-01-10 by reaper7