reaper7/SDM_Energy_Meter #36: Writing to / Configering the SDM120

deepthroatme, 2019-10-13:
Dear repaer7, 

I like to use 3 SDM120 on one EIA-485 bus. For this I have to change the Meter ID. And I also like to set another Baudrate and parity. I have found no e.g. "sdm.write" in you library. 

How long do you need to extend your library? And how much should I spend via PayPal for your work? 

Kind regards

reaper7, 2019-10-13:
what do you want to extend?
Baudrate and parity You can set, this is described in INITIALIZING section:
https://github.com/reaper7/SDM_Energy_Meter#initializing

Reading from meters with differend id's is described in READING section:
https://github.com/reaper7/SDM_Energy_Meter#reading

all meters must be set with the same baudrate and parity, only id's must be different for each meter

deepthroatme, 2019-10-13:
Thanks for you fast answer! I have read and (think) understand the INITIALIZING and READING section. 

<<< all meters must be set with the same baudrate and parity, only id's must be different for each meter >>> Yes but how can I set this? It seems I have to press the button on the SDM120 for some time an then I can set the parameters ONLY by software and then reset the SDM120... 

Sorry if I to stupid for your SW

reaper7, 2019-10-14:
Can not be set with the button? I don't know, I don't have this model.

This library does not have this functionality and currently I'm not planning implementation.
For a one-time procedure? Really?
You can do it faster using a computer, just use this pythons project for this:
https://github.com/gianfrdp/SDM120C

deepthroatme, 2019-10-14:
<<< Can not be set with the button? >>> SDM120 Not at this moment, but I will ask the producer / importeur... 

<<< For a one-time procedure? Really? >>> Yes :-( 

<<< You can do it faster using a computer, just use for e.g. this project: https://github.com/gianfrdp/SDM120C >>>  Thanks for your help! 
But I dont understand Phyton and how to you use it :-( 
So I will try out other Arduino/ESP8266 SDM120 librarys....

Sorry to disturb you!

reaper7, 2019-10-14:
ok

closed 2019-10-14 by reaper7