reaper7/SDM_Energy_Meter #39: sdm compiling

Giulio1980, 2019-10-19:
Good morning,
I was happily using old version of the library for a year and a half.
Unfortunately I updated the ARDUINO IDE, the ESP8266 library to 2.5.2 and doing so the Softwareserial that was working on pin 4,5.

Now, I tryed to change the lines in my project and it doesn't work.

I updated the library, let everything as it is (I mean I didn't even change the pins in the sdmconfigurator) but even just trying the sdm_simple project:
- If I use the Hardware definition it compile fine. 
- I use the software serial I can't compile 
"
sdm_simple:36:16: error: no matching function for call to 'SoftwareSerial::SoftwareSerial()'

 SoftwareSerial swSerSDM;                                                        //config SoftwareSerial
"
I tried to check on google for such a matter but I just can't resolve it.

Thank you in advance
Loris

reaper7, 2019-10-19:
Probably you have an older version of SoftwareSerial lib.
Check in your adruino esp directory > esp8266\libraries\SoftwareSerial in library.properties files.
The situation is now complicated because in the old version of swserial library
parameters were passed in the constructor but in the new version they have been moved to begin.

I use the arduino esp current git version and for this reason also a swserial library in the latest version 5.3.4 and my library is adapted to this.

you have two solutions:
if you have an old version of the software serial then you must also use my older library version 2.0.3,
which you will find here: https://github.com/reaper7/SDM_Energy_Meter/releases

if you have the latest version of the software serial you must also have the latest version of my library 2.0.5

Giulio1980, 2019-10-19:
Very kind of you to answer.
- Eventually I changed the cable to work with hardware serial port... I check if this way the board is more stable.
- and thanks to your answer I check the library and I had 5.0.4 (download in the last version of the package, maybe it's time to update the package) and it's now working.

I'll keep it this way for now, and I'll make experiment when I will have the real necessity for two serial.
Thank you very much

p.s. I don't know what are the rules, but it would have saved me two days of experimenting if there was written the version of software this new library was working with.  Or maybe it is written somewhere...

reaper7, 2019-10-19:
as I wrote many times, I personally use only hwserial...

about the change in the swserial library I learned by watching the changes in the arduino esp repository:
https://github.com/esp8266/Arduino/commit/122e87019f35ecb928c00631b6dbd4d581e12105
and decided to improve my library...
by the way I reported one discrepancy:
https://github.com/plerup/espsoftwareserial/issues/117
and one issue:
https://github.com/plerup/espsoftwareserial/issues/118

reaper7, 2019-10-19:
I added information about the required version:
https://github.com/reaper7/SDM_Energy_Meter#introduction

Please close this issue if everything is alright

closed 2019-10-19 by Giulio1980