reaper7/SDM_Energy_Meter #40: Slave ID not working.

HilverinkJ, 2019-11-05:
Hi,

I do have 3 slaves on the same RS485 line.

Reading SDM power with ESP32 from one divice is working fine.
`float power1 = sdm.readVal(SDM220T_POWER);`

`float power2 = sdm.readVal(SDM220T_POWER, 0x02);`
results in nan.

Your example sdm_simple:

 ` Serial.print("Power:     ");`
  `Serial.print(sdm.readVal(SDM220T_POWER), 2);                                  //display power`
  `Serial.println("W");`
Is working fine but only for one slave as defined with:

`#define SDM_B_01                            0x01                                //BYTE 1 -> slave address (default value 1 read from node 1)`

Regards,
Joop.

reaper7, 2019-11-05:
I understand you have also changed the addresses in the meters?
if so, leave only on the line the meter with ID 2 and try to read:
```
float power2 = sdm.readVal(SDM220T_POWER, 0x02);
```

reaper7, 2019-12-04:
closed due to inactivity

closed 2019-12-04 by reaper7