reaper7/SDM_Energy_Meter #43: SDM_energy_meter library adapter for ORNO_energy_meter

oussama415, 2020-03-31:
Hello, please I would like to adapt the SDM energy-meter library for my OR-WE-517 meter,
I downloaded it, I modified the SDM.h file by modifying the addresses which correspond to the quantities that I want to measure
this is SDM.h :

> SDM.h
`
//------------------------------------------------------------------------------
#ifndef SDM_h
#define SDM_h
//------------------------------------------------------------------------------
#include <Arduino.h>
#include "SDM_Config_User.h"
#if defined ( USE_HARDWARESERIAL )
  #include <HardwareSerial.h>
#else
  #include <SoftwareSerial.h>
#endif
//------------------------------------------------------------------------------
//DEFAULT CONFIG (DO NOT CHANGE ANYTHING!!! for changes use SDM_Config_User.h):
//------------------------------------------------------------------------------
#if !defined ( SDM_UART_BAUD )
  #define SDM_UART_BAUD                     9600                               //default baudrate 
#endif

#if !defined ( DERE_PIN )
  #define DERE_PIN                          NOT_A_PIN                           //default digital pin for control MAX485 DE/RE lines (connect DE & /RE together to this pin)
#endif

#if defined ( USE_HARDWARESERIAL )

  #if !defined ( SDM_UART_CONFIG )
    #define SDM_UART_CONFIG                 SERIAL_8N1                          //default hardware uart config 
  #endif

  #if defined ( ESP8266 ) && !defined ( SWAPHWSERIAL )
    #define SWAPHWSERIAL                    0                                   //(only esp8266) when hwserial used, then swap uart pins from 3/1 to 13/15 (default not swap)
  #endif

  #if defined ( ESP32 )
    #if !defined ( SDM_RX_PIN )
      #define SDM_RX_PIN                    -1                                  //use default rx pin for selected port
    #endif
    #if !defined ( SDM_TX_PIN )
      #define SDM_TX_PIN                    -1                                  //use default tx pin for selected port
    #endif
  #endif

#else

  #if defined ( ESP8266 ) || defined ( ESP32 )
    #if !defined ( SDM_UART_CONFIG )
      #define SDM_UART_CONFIG               SWSERIAL_8N1                        //default softwareware uart config for esp8266/esp32 :::::::
    #endif
  #endif

//  #if !defined ( SDM_RX_PIN ) || !defined ( SDM_TX_PIN )
//    #error "SDM_RX_PIN and SDM_TX_PIN must be defined in SDM_Config_User.h for Software Serial option)"
//  #endif

  #if !defined ( SDM_RX_PIN )
    #define SDM_RX_PIN                      -1
  #endif
  #if !defined ( SDM_TX_PIN )
    #define SDM_TX_PIN                      -1
  #endif

#endif

#if !defined ( MAX_MILLIS_TO_WAIT )
  #define MAX_MILLIS_TO_WAIT                500                                 //default max time to wait for response from SDM !!!!
[... 123 lines cut ...]

  delay(50);
  
  Serial.print("Power:     ");
  Serial.print(sdm.readVal(ORNO_FREQUENCY ), ID_note);                                  
  Serial.println("W");
   delay(50);

  Serial.print(sdm.getErrCode());

  delay(1000);                                                                  
}`

this is an example of monitoring for my energy-meter, baudrait is 9600, Even, 8, 1

[image]

i get always nan value on serial monitor 

`Voltage:   nanV
Current:   nanA
Power:     nanW
4..............................
`
Can you help me please !

reaper7, 2020-04-01:
Sorry but I can't help you too much...

look at the frame you send using the original program (WRITE):
[image]

and compare with this:
https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.cpp#L82

first byte is a node id and looks good == 0x01,

third and fourth byte looks good too, because they specify the register number we want to read (0x00, 0x0e),

fifth and sixth byte looks good too, your program and this library sends 0x00 and 0x02

**but note the second byte ! ! !**
My lib send 0x04 value at this position (read from 3X input registers):
https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.h#L76

but Your program reads values from another register (holding registers),
and send 0x03 value as second byte...

so...try to change SDM_B_02 value from 0x04 to 0x03 at:
https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.h#L76

oussama415, 2020-04-01:
i think i have a problem with configuration how can i modify configuration like this 
[image]

reaper7, 2020-04-01:
in SDM_Config_User.h
uncomment and / or modify the appropriate values in the right places
depending on whether you use hardware or software serial

```
#define SDM_UART_BAUD                       9600

#if defined ( USE_HARDWARESERIAL )
#define SDM_UART_CONFIG                   SERIAL_8E1         //if hardware serial (parity EVEN)
#else
#define SDM_UART_CONFIG                   SWSERIAL_8E1   //if software serial (parity EVEN)
#endif
```

reaper7, 2020-04-23:
closed due to inactivity

closed 2020-04-23 by reaper7

max321321231, 2020-05-02:
hey @oussama415 ,

I have the same problem as you,
did you find a solution?

CSchoch, 2020-06-18:
I've done that already. You can check out:
https://github.com/CSchoch/OR_WE_Energy_Meter