reaper7/SDM_Energy_Meter #46: Question about Error Codes - Meaning/Frequency

Timmmy001, 2020-06-01:
I have Questions which codes are of concern and in which frequency are negligible:
With 26815 readings (with 24 reading every 10 sec. / approx 17h long) I get 33 errors each code 4 (timeout?).
I use HardwareSerial on a ESP32.

Stupid question, but is this a good average or in need of improvement?

reaper7, 2020-06-01:
defenitely too much! personally, I didn't use or check esp32
but I will try to check...
anyway I also use hw serial at 9600 (with esp8266) and I don't have any errors.
8 days without error...
[image]

try to check the connections, cable quality, power supply...

Timmmy001, 2020-06-01:
Power on 3,3V-PIN is ok (supply Modbus-Module too)
Powered from USB-Charger (4,87V under load)

I use standard parameters:
`SDM sdm(Serial, 9600, NOT_A_PIN, SERIAL_8N1, 16, 17);`

From what source do you get "TS Write Error Codes"?

Back to Error code 4:
What mean this? Timeout? From my Modbus-Module/ESP (too slow/stuck bits)?

reaper7, 2020-06-01:
do not pay attention to TS... this is my sketch which loads results to thingspeak! (and this is http error list)

so... I prepared another example, this time for esp32:
https://github.com/reaper7/SDM_Energy_Meter/tree/master/examples/sdm_live_page_esp32_hwserial

and at the moment I am testing for possible errors...
I won't be able to do it long enough because my home automation needs live data from my orginal sketch

in the example given above, adjust the pins to your configuration at line:
https://github.com/reaper7/SDM_Energy_Meter/blob/master/examples/sdm_live_page_esp32_hwserial/sdm_live_page_esp32_hwserial.ino#L43

Error code 4 is **timeout**: (https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.h#L224)
and means, that in the time declared by MAX_MILLIS_TO_WAIT (default 500ms)  (https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM_Config_User.h#L84)
**esp** has not received the complete data from your meter.
This piece of code starts in a line: https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.cpp#L110
**esp** in a loop checks the uart receive buffer until the complete 9 bytes are received
or if MAX_MILLIS_TO_WAIT time is exceeded.
In your case the second situation occurs...

At the beginning
Eliminate the first visible problem, lack of full 5V voltage and change the power supply to a more powerful...it doesn't have to be, but it can matter.

my esp32 after half hour (~4200readings without error)  
[image]
reading 5 registers every 2sec

reaper7, 2020-06-26:
closed due to inactivity

closed 2020-06-26 by reaper7