reaper7/SDM_Energy_Meter #48: Wemos D1 mini after HW reset - problem with serial

Chris125913, 2020-11-13:
Hi Everyone,
Hi reaper7 ,
I use your code to read some values from SDM72D-M. Everything works fine but after HW reset (or power off) program hangs out and transmission was stopped. Only TX led on RS458 converter lights on. The only solution I've found is to disconnect software serial D8/D7 lines from Wemod, reset Wemod, and after that connect this line. In my tests, I used the *.ino file
https://github.com/reaper7/SDM_Energy_Meter/blob/master/examples/sdm_simple/sdm_simple.ino
modified to read only one value from SDM72.

I have tested this at two Wemos boards.

Does someone have a similar problem? 
Thanks for your help

PS
This my first post on GitHub so I think I didn't break any rules.

reaper7, 2020-11-13:
Did You read Initializing section of readme (note at the end of this section)?
https://github.com/reaper7/SDM_Energy_Meter#initializing
GPIO15 (D8) at start must be at LOW state  (https://www.forward.com.au/pfod/ESP8266/GPIOpins/index.html)
this is a ESP8266 requirement, but probably Your converter has pull-up resistors on TX/RX lines,
therefore it is not possible to start ESP8266 correctly.

Chris125913, 2020-11-13:
Thanks for fast answer. I don’t how but I don’t see this note. Thanks for help. I will check it. 

Wysłane z iPhone'a

Chris125913, 2020-11-17:
Hi all. I checked this and I found my bug. You have the right reaper7. Now my Wemod is collecting data from three sdm meters and all working perfectly. THX a lot

reaper7, 2020-11-18:
@Chris125913 - I'm always right...with a few exceptions, when I'm wrong ;) 
so...please close this issue
Good luck!

closed 2020-11-18 by Chris125913