reaper7/SDM_Energy_Meter #49: FYI: weird stop bit behavior

MoritzBiester, 2020-11-18:
Hi there!

First off all. Thank you for publishing your code here.
I used it to read out my SDM630 Modbus V2 in combination with an influxDB.

I ran in some issues when using the following settings:
SDM630: 2400baud, no parity, 1 stopbit

As you can see in the following picture the last stopbit of the response is out of spec. (only sometimes).
[link to the unknown ;)](https://s12.directupload.net/images/201118/i5y4b2kg.png)

Fixed it by using 2 stop bits.

Have a nice day
M.Biester

reaper7, 2020-11-18:
@MoritzBiester - 8N1 is a default settings for this lib and from the beginning I have been using such settings without any problem.
1. you use hw or sw implementation?
2. your oscillogram shows signals on A B pins (converter output) but what signals looks on converter input (on tx/rx uC pins)?
3. oscillogram shows transmission from uC to SDM or from SDM to uC?

Basically this library does not affect the transmission, because it doesn't use its own implementations.
It uses builtin serial.write

MoritzBiester, 2020-11-18:
1. I was using Softwareserial 8N1(ESP and SDM630), changed it to 8N2(ESP & SDM630)
2. the rx pin on the ESP8266 corresponds 
3. Shows the Modbus lines(A, B) with the request(first 8 bytes) and the response(last 9 bytes and crippled stopbit)

The software is all working fine. 
Its just some weirdness from the SDM630 releasing its dere pin to early or something.
Maybe the implemented a pourly designed auto direction control on their RS485 transceiver IC.

This was just for information purpose. So if someone else runs into this issue, they dont have to spend  a whole day debugging in the cold dark basement.
(silly me had it installed before he tested it)

It all works with no errors since two days.(36000 measurements).

closed 2020-11-18 by MoritzBiester

reaper7, 2020-11-19:
Interesting information.
I will pin this thread to the problems section.