reaper7/SDM_Energy_Meter #50: SDM 72D reboots if Modbus reads are too fast

DGrebe, 2020-12-14:
Hi reaper7

Thanks for this great library and the examples.

I am working with SDM 120C & SDM72D. With the SDM 120C requests can be done without any delay. When I try the same with SDM72D the energy meter reboots after 1 or 2 requests, so that the transmission of more than 1 measure in the read intervall does not work stable. I got it stabilized by putting a delay of 500 ms after the sdm.readVal request.

I am using the example "sdm_live_page_esp8266_hwserial.ino" with a swapped hwSerial on pins 13/15. I had made the same experiences with ESPeasy using the SDM plugin (testing)

Has anybody else made similar experiences or is something wrong with the energy meter SDM72D.
In my experimental setup I have just connected L1 and N to 230V. I have not tried it with 3 phases.

I want to bring ESPeasy to work with the SDM 72D and are willing to port the new SDM library to the ESPeasy framework.
So far it is still using the old SDM lib with the separate constants for the different energy meters.

Regards

Dirk

[sdm_live_page_esp8266_hwserial.ino.txt](https://github.com/reaper7/SDM_Energy_Meter/files/5691549/sdm_live_page_esp8266_hwserial.ino.txt)

reaper7, 2020-12-16:
this is very strange behavior in a professional device...even if the next query comes too early, it should answer with an error or not answer at all but not reboot itself ...
Generally your observations are correct....
Modern devices have no problem with faster reading of subsequent parameters,
but in the assumptions of the modbus protocol there are time requirements for subsequent readings.

readVal procedure is assumed to read one value,
so i won't add any delay there
however, I can improve the examples to add a delay even in the loop reading.
but what delay value? 500ms? 1000ms :)  my SDM220 is read in a loop without any delay and it doesn't even cause errors and reboots...

Please read this:
https://ctlsys.com/support/modbus_message_timing/

one thing is for sure - this library does not contain the correct modbus protocol implementation 
was designed to read my meter and it does it properly :D

reaper7, 2020-12-16:
Try current master,
a definable parameter RESPONSE_TIMEOUT has been added (default 500ms)
This value can be changed by the user in the SDM_Config_User.h file

closed 2021-01-13 by reaper7

DGrebe, 2021-01-13:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAEgAAABICAYAAABV7bNHAAAABHNCSVQICAgIfAhkiAAAACtJREFUeJztwQENAAAAwqD3T20ON6AAAAAAAAAAAAAAAAAAAAAAAAAAAD4MUUgAARy2AfAAAAAASUVORK5CYII=###24x24:true###][Tracking-Bild][https://github.com/notifications/beacon/AA3RDHPBYI7K4QJ3UHNEZFTSZU3ALA5CNFSM4U3KMLQ2YY3PNVWWK3TUL52HS4DFWZEXG43VMVCXMZLOORHG65DJMZUWGYLUNFXW5KTDN5WW2ZLOORPWSZGO7JIB6QY.gif]

Thanks!