reaper7/SDM_Energy_Meter #51: No Reads from SDM630

Darnshelm, 2021-01-12:
Hi
I totaly new to esp and modbus. I have try to fallow your instructions but i miss something I don't get any read value from 3 SDM630. I have use our sdm_live_page_esp8266_hwserial.
I get the esp8266 to boot and i can access the webpage.
I have same converter 
I have connected the converter to pin 3/1 on esp8266.
convertar are connected A-A B-B and gnd 2-
I have read many close issues to see if i can understand and find then problem but i have not.
I try shift cabel inpus from sdm630
try change stopbit.
hope it easy to solve

THX 
//David

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

reaper7, 2021-01-13:
@Darnshelm - show what looks your **SDM_Config_User.h** and complete sketch file.

Do you have connected one or three meters ?
If three then leave only one, until you solve the communication problem.

What about device addresses? I can see on the meter that you have set address 3, did you include it in the code??

Darnshelm, 2021-01-13:
HI
I have connected 3 Meters
Yes I have set the adress on all 3 meters. Main meter to 1 garage to 2 House to 3
no I didn't include the adress in the code.
it only blinks on TXD
Yes it sends 8 bits 01 04 00 00 00 02 71 CB 

[SDM_Config_User.txt](https://github.com/reaper7/SDM_Energy_Meter/files/5808696/SDM_Config_User.txt)

reaper7, 2021-01-13:
uncomment line 17 in SDM_Config_User.txt
from:
//#define USE_HARDWARESERIAL
to:
#define USE_HARDWARESERIAL

Unmodified sketch trying to read from meter with addr 001,
so on this meter (with address 001) set the correct transmission parameters (baudrate, parity, stop bits) like in a sketch,
disconnect the other two meters (002, 003) and try again

Darnshelm, 2021-01-13:
Hi thx for you help now it works with one meter.
but i just see only one phase V and AMPs
what lines should i uncomment to see more

and how should the lines for the another 2 meters lock like?

reaper7, 2021-01-13:
You must expand code not uncoment something :D
both ino and index_page.h as well

Darnshelm, 2021-01-13:
Okay 
it there a example where i can copy and paste from?

reaper7, 2021-01-13:
please... :D

I see that you can analyze and solve problems on your own, you can do it yourself...

reaper7, 2021-01-13:
**sdm630_multi_live_page_esp8266_hwserial.ino**
```cpp
//sdm live page example by reaper7

#define READSDMEVERY  2000                                                      //read sdm every 2000ms
#define NBREG   39                                                              //number of sdm registers to read
//#define USE_STATIC_IP

/*  WEMOS D1 Mini
                     ______________________________
                    |   L T L T L T L T L T L T    |
                    |                              |
                 RST|                             1|TX HSer
                  A0|                             3|RX HSer
                  D0|16                           5|D1
                  D5|14                           4|D2
                  D6|12                    10kPUP_0|D3
RX SSer/HSer swap D7|13                LED_10kPUP_2|D4
TX SSer/HSer swap D8|15                            |GND
                 3V3|__                            |5V
                       |                           |
                       |___________________________|
*/

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>

#include <ESPAsyncTCP.h>                                                        //https://github.com/me-no-dev/ESPAsyncTCP
#include <ESPAsyncWebServer.h>                                                  //https://github.com/me-no-dev/ESPAsyncWebServer

#include <SDM.h>                                                                //https://github.com/reaper7/SDM_Energy_Meter

#include "index_page.h"

#if !defined ( USE_HARDWARESERIAL )
  #error "This example works with Hardware Serial on esp8266, please uncomment #define USE_HARDWARESERIAL in SDM_Config_User.h"
#endif
//------------------------------------------------------------------------------
AsyncWebServer server(80);

SDM sdm(Serial, 9600, NOT_A_PIN, SERIAL_8N1, false);                            //HARDWARE SERIAL

//------------------------------------------------------------------------------
String devicename = "PWRMETER";

#if defined ( USE_STATIC_IP )
IPAddress ip(192, 168, 0, 130);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 255, 0);
#endif

const char* wifi_ssid = "YOUR_WIFI_SSID";
const char* wifi_password = "YOUR_WIFI_PASS";

String lastresetreason = "";

unsigned long readtime;
//------------------------------------------------------------------------------
typedef volatile struct {
  const uint8_t meter;
  volatile float regvalarr;
  const uint16_t regarr;
} sdm_struct;

volatile sdm_struct sdmarr[NBREG] = {
  {1, 0.00, SDM_PHASE_1_VOLTAGE},                                               //V
  {1, 0.00, SDM_PHASE_2_VOLTAGE},                                               //V
  {1, 0.00, SDM_PHASE_3_VOLTAGE},                                               //V
[... 237 lines cut ...]
        <TR><TH title="POWER FACTOR">POWER FACTOR PH2 M2</TH><TD><A id='resp23'></A></TD><TD>PF</TD></TR>
        <TR><TH title="POWER FACTOR">POWER FACTOR PH3 M2</TH><TD><A id='resp24'></A></TD><TD>PF</TD></TR>
        <TR><TH title="FREQUENCY">FREQUENCY M2</TH><TD><A id='resp25'></A></TD><TD>Hz</TD></TR>
        <TR><TH title="VOLTAGE">VOLTAGE PH1 M3</TH><TD><A id='resp26'></A></TD><TD>V</TD></TR>
        <TR><TH title="VOLTAGE">VOLTAGE PH2 M3</TH><TD><A id='resp27'></A></TD><TD>V</TD></TR>
        <TR><TH title="VOLTAGE">VOLTAGE PH3 M3</TH><TD><A id='resp28'></A></TD><TD>V</TD></TR>
        <TR><TH title="CURRENT">CURRENT PH1 M3</TH><TD><A id='resp29'></A></TD><TD>A</TD></TR>
        <TR><TH title="CURRENT">CURRENT PH2 M3</TH><TD><A id='resp30'></A></TD><TD>A</TD></TR>
        <TR><TH title="CURRENT">CURRENT PH3 M3</TH><TD><A id='resp31'></A></TD><TD>A</TD></TR>
        <TR><TH title="POWER">POWER PH1 M3</TH><TD><A id='resp32'></A></TD><TD>W</TD></TR>
        <TR><TH title="POWER">POWER PH2 M3</TH><TD><A id='resp33'></A></TD><TD>W</TD></TR>
        <TR><TH title="POWER">POWER PH3 M3</TH><TD><A id='resp34'></A></TD><TD>W</TD></TR>
        <TR><TH title="POWER FACTOR">POWER FACTOR PH1 M3</TH><TD><A id='resp35'></A></TD><TD>PF</TD></TR>
        <TR><TH title="POWER FACTOR">POWER FACTOR PH2 M3</TH><TD><A id='resp36'></A></TD><TD>PF</TD></TR>
        <TR><TH title="POWER FACTOR">POWER FACTOR PH3 M3</TH><TD><A id='resp37'></A></TD><TD>PF</TD></TR>
        <TR><TH title="FREQUENCY">FREQUENCY M3</TH><TD><A id='resp38'></A></TD><TD>Hz</TD></TR>
        <TR><TH title="FREE HEAP">FREE HEAP</TH><TD><A id='freeheap'></A></TD><TD>bytes</TD></TR>
        <TR><TH title="LAST RESET REASON">LAST RESET REASON</TH><TD colspan="2"><A id='rstreason'></A></TD></TR>
      </TABLE>
    </CENTER>
  </BODY>
</HTML>
)=====";

```

Darnshelm, 2021-01-13:
thanks you very much for all help and for the codes and the motivation. had started and found what lines and what code I needed.
it does not have to look good, just download data from there to the home assistant.

reaper7, 2021-01-13:
You need to work on the appearance of the html page in the index_page.h file ... you know ... tables, rows, columns

for HA I use esphome with custom component:
https://github.com/reaper7/esphome_sdm_custom
of course it also needs to be adapted to your needs

reaper7, 2021-01-13:
little better page:

**index_page.h:**
```html
const char index_page[] PROGMEM = R"=====(
<!DOCTYPE HTML>
<HTML>
  <HEAD>
    <META name='viewport' content='width=device-width, initial-scale=1'>
    <TITLE>SDM live POWER table</TITLE>
    <SCRIPT>
    var xmlHttp=createXmlHttpObject();
    function createXmlHttpObject(){
     if(window.XMLHttpRequest){
        xmlHttp=new XMLHttpRequest();
     }else{
        xmlHttp=new ActiveXObject('Microsoft.XMLHTTP');
     }
     return xmlHttp;
    }
    function process(){
      if(xmlHttp.readyState==0 || xmlHttp.readyState==4){
        xmlHttp.open('PUT','xml',true);
        xmlHttp.onreadystatechange=handleServerResponse;
        xmlHttp.send(null);
      }
      setTimeout('process()',2000);
    }
    function handleServerResponse(){
     if(xmlHttp.readyState==4 && xmlHttp.status==200){
       xmlResponse=xmlHttp.responseXML;
       for(i=0;i<39;i++){
        xmldoc=xmlResponse.getElementsByTagName('response'+i)[0].firstChild.nodeValue;
        document.getElementById('resp'+i).innerHTML=xmldoc;
       }
       xmldoc=xmlResponse.getElementsByTagName('freeh')[0].firstChild.nodeValue;
       document.getElementById('freeheap').innerHTML=xmldoc;
       xmldoc=xmlResponse.getElementsByTagName('rst')[0].firstChild.nodeValue;
       document.getElementById('rstreason').innerHTML=xmldoc;
     }
    }
    </SCRIPT>
    <STYLE>
      h1 {
        font-size: 120%;
        color: blue;
        margin: 0 0 10px 0;
      }
       table{
        border-collapse: collapse;
      }     
      table, th, td {
        text-align: center;
        border: 1px solid blue;
      }
      tr:nth-child(even) {background-color: #f2f2f2}
    </STYLE>
  </HEAD>
  <BODY onload='process()'>
    <CENTER>
      <H1>SDM live POWER table</H1>
      <TABLE BORDER=1>
        <TR><TH rowspan="5", title="METER 1">METER 1</TH><TD>VOLTAGE</TD><TD><A id='resp0'></A></TD><TD><A id='resp1'></A></TD><TD><A id='resp2'></A></TD><TD>V</TD></TR>
        <TR><TD>CURRENT</TD><TD><A id='resp3'></A></TD><TD><A id='resp4'></A></TD><TD><A id='resp5'></A></TD><TD>A</TD></TR>
        <TR><TD>POWER</TD><TD><A id='resp6'></A></TD><TD><A id='resp7'></A></TD><TD><A id='resp8'></A></TD><TD>W</TD></TR>
        <TR><TD>POWER FACT.</TD><TD><A id='resp9'></A></TD><TD><A id='resp10'></A></TD><TD><A id='resp11'></A></TD><TD>PF</TD></TR>
        <TR><TD>FREQUENCY</TD><TD colspan="3"><A id='resp12'></A></TD><TD>Hz</TD></TR>
      </TABLE>
      <BR/>
      <TABLE BORDER=1>
        <TR><TH rowspan="5", title="METER 2">METER 2</TH><TD>VOLTAGE</TD><TD><A id='resp13'></A></TD><TD><A id='resp14'></A></TD><TD><A id='resp15'></A></TD><TD>V</TD></TR>
        <TR><TD>CURRENT</TD><TD><A id='resp16'></A></TD><TD><A id='resp17'></A></TD><TD><A id='resp18'></A></TD><TD>A</TD></TR>
        <TR><TD>POWER</TD><TD><A id='resp19'></A></TD><TD><A id='resp20'></A></TD><TD><A id='resp21'></A></TD><TD>W</TD></TR>
        <TR><TD>POWER FACT.</TD><TD><A id='resp22'></A></TD><TD><A id='resp23'></A></TD><TD><A id='resp24'></A></TD><TD>PF</TD></TR>
        <TR><TD>FREQUENCY</TD><TD colspan="3"><A id='resp25'></A></TD><TD>Hz</TD></TR>
      </TABLE>
      <BR/>
      <TABLE BORDER=1>
        <TR><TH rowspan="5", title="METER 2">METER 2</TH><TD>VOLTAGE</TD><TD><A id='resp26'></A></TD><TD><A id='resp27'></A></TD><TD><A id='resp28'></A></TD><TD>V</TD></TR>
        <TR><TD>CURRENT</TD><TD><A id='resp29'></A></TD><TD><A id='resp30'></A></TD><TD><A id='resp31'></A></TD><TD>A</TD></TR>
        <TR><TD>POWER</TD><TD><A id='resp32'></A></TD><TD><A id='resp33'></A></TD><TD><A id='resp34'></A></TD><TD>W</TD></TR>
        <TR><TD>POWER FACT.</TD><TD><A id='resp35'></A></TD><TD><A id='resp36'></A></TD><TD><A id='resp37'></A></TD><TD>PF</TD></TR>
        <TR><TD>FREQUENCY</TD><TD colspan="3"><A id='resp38'></A></TD><TD>Hz</TD></TR>
      </TABLE>
      <BR/>
      <TABLE BORDER=1>
        <TR><TH rowspan="2", title="ESP">ESP</TH><TD>FREE HEAP</TD><TD><A id='freeheap'></A></TD><TD>bytes</TD></TR>
        <TR><TD>FREQUENCY</TD><TD colspan="2"><A id='rstreason'></A></TD></TR>
      </TABLE>
    </CENTER>
  </BODY>
</HTML>
)=====";

```

Darnshelm, 2021-01-16:
Hi
Thx yes the site locks much better.
I try to find where to tell esp8266 it 3 power meters to read of.
I copy and past your code and i get the nice tabel on webpage. 
 also use the code lines on sdm_live_page you send.
but I only get it to read from one meter only

reaper7, 2021-01-17:
if You connect esp to PC (as we did before), you should have such inquiries on terminal:
```
01 04 00 00 00 02 71 cb  ......qË
01 04 00 02 00 02 d0 0b  ......Đ.
01 04 00 04 00 02 30 0a  ......0.
01 04 00 06 00 02 91 ca  ......‘Ę
01 04 00 08 00 02 f0 09  ......đ.
01 04 00 0a 00 02 51 c9  ......QÉ
01 04 00 0c 00 02 b1 c8  ......±Č
01 04 00 0e 00 02 10 08  ........
01 04 00 10 00 02 70 0e  ......p.
01 04 00 1e 00 02 11 cd  .......Í
01 04 00 20 00 02 70 01  ... ..p.
01 04 00 22 00 02 d1 c1  ..."..ŃÁ
01 04 00 46 00 02 90 1e  ...F...
02 04 00 00 00 02 71 f8  ......qř
02 04 00 02 00 02 d0 38  ......Đ8
02 04 00 04 00 02 30 39  ......09
02 04 00 06 00 02 91 f9  ......‘ů
02 04 00 08 00 02 f0 3a  ......đ:
02 04 00 0a 00 02 51 fa  ......Qú
02 04 00 0c 00 02 b1 fb  ......±ű
02 04 00 0e 00 02 10 3b  .......;
02 04 00 10 00 02 70 3d  ......p=
02 04 00 1e 00 02 11 fe  .......ţ
02 04 00 20 00 02 70 32  ... ..p2
02 04 00 22 00 02 d1 f2  ..."..Ńň
02 04 00 46 00 02 90 2d  ...F..-
03 04 00 00 00 02 70 29  ......p)
03 04 00 02 00 02 d1 e9  ......Ńé
03 04 00 04 00 02 31 e8  ......1č
03 04 00 06 00 02 90 28  ......(
03 04 00 08 00 02 f1 eb  ......ńë
03 04 00 0a 00 02 50 2b  ......P+
03 04 00 0c 00 02 b0 2a  ......°*
03 04 00 0e 00 02 11 ea  .......ę
03 04 00 10 00 02 71 ec  ......qě
03 04 00 1e 00 02 10 2f  ......./
03 04 00 20 00 02 71 e3  ... ..qă
03 04 00 22 00 02 d0 23  ..."..Đ#
03 04 00 46 00 02 91 fc  ...F..‘ü
```
first bytes of each query (next lines) contain addresses of meters to which inquiries are addressed.
As you can see, at the beginning, the meter with the address 01 is read (all the data that interests us),
then the meter with the address 02 is queried for the same values, and finally the meter with the address 03

The procedure that queries the meters is:
```cpp
void sdmRead() {
  float tmpval = NAN;
  for (uint8_t i = 0; i < NBREG; i++) {
    tmpval = sdm.readVal(sdmarr[i].regarr,sdmarr[i].meter);
    if (isnan(tmpval))
      sdmarr[i].regvalarr = 0.00;
    else
      sdmarr[i].regvalarr = tmpval;

    yield();
  }
}
```
the query is described in the documentation:
https://github.com/reaper7/SDM_Energy_Meter#reading
where the last (second) parameter specifies the address of the meter being queried.

What parameters and from which meter we will take is specified by this "list":
```cpp
volatile sdm_struct sdmarr[NBREG] = {
  {1, 0.00, SDM_PHASE_1_VOLTAGE},                                               //V
  {1, 0.00, SDM_PHASE_2_VOLTAGE},                                               //V
  {1, 0.00, SDM_PHASE_3_VOLTAGE},                                               //V
  {1, 0.00, SDM_PHASE_1_CURRENT},                                               //A
  {1, 0.00, SDM_PHASE_2_CURRENT},                                               //A
  {1, 0.00, SDM_PHASE_3_CURRENT},                                               //A
  {1, 0.00, SDM_PHASE_1_POWER},                                                 //W
  {1, 0.00, SDM_PHASE_2_POWER},                                                 //W
  {1, 0.00, SDM_PHASE_3_POWER},                                                 //W
  {1, 0.00, SDM_PHASE_1_POWER_FACTOR},                                          //PF
  {1, 0.00, SDM_PHASE_2_POWER_FACTOR},                                          //PF
  {1, 0.00, SDM_PHASE_3_POWER_FACTOR},                                          //PF
  {1, 0.00, SDM_FREQUENCY},                                                     //Hz
  {2, 0.00, SDM_PHASE_1_VOLTAGE},                                               //V
  {2, 0.00, SDM_PHASE_2_VOLTAGE},                                               //V
  {2, 0.00, SDM_PHASE_3_VOLTAGE},                                               //V
  {2, 0.00, SDM_PHASE_1_CURRENT},                                               //A
  {2, 0.00, SDM_PHASE_2_CURRENT},                                               //A
  {2, 0.00, SDM_PHASE_3_CURRENT},                                               //A
  {2, 0.00, SDM_PHASE_1_POWER},                                                 //W
  {2, 0.00, SDM_PHASE_2_POWER},                                                 //W
  {2, 0.00, SDM_PHASE_3_POWER},                                                 //W
  {2, 0.00, SDM_PHASE_1_POWER_FACTOR},                                          //PF
  {2, 0.00, SDM_PHASE_2_POWER_FACTOR},                                          //PF
  {2, 0.00, SDM_PHASE_3_POWER_FACTOR},                                          //PF
  {2, 0.00, SDM_FREQUENCY},                                                     //Hz
  {3, 0.00, SDM_PHASE_1_VOLTAGE},                                               //V
  {3, 0.00, SDM_PHASE_2_VOLTAGE},                                               //V
  {3, 0.00, SDM_PHASE_3_VOLTAGE},                                               //V
  {3, 0.00, SDM_PHASE_1_CURRENT},                                               //A
  {3, 0.00, SDM_PHASE_2_CURRENT},                                               //A
  {3, 0.00, SDM_PHASE_3_CURRENT},                                               //A
  {3, 0.00, SDM_PHASE_1_POWER},                                                 //W
  {3, 0.00, SDM_PHASE_2_POWER},                                                 //W
  {3, 0.00, SDM_PHASE_3_POWER},                                                 //W
  {3, 0.00, SDM_PHASE_1_POWER_FACTOR},                                          //PF
  {3, 0.00, SDM_PHASE_2_POWER_FACTOR},                                          //PF
  {3, 0.00, SDM_PHASE_3_POWER_FACTOR},                                          //PF
  {3, 0.00, SDM_FREQUENCY}                                                      //Hz
};
```
where the meter address comes first on each line.

So you have to make sure that the meters have addresses of 001, 002 and 003 respectively.

reaper7, 2021-04-26:
lack of activity. closed

closed 2021-04-26 by reaper7