reaper7/SDM_Energy_Meter #52: Flow direction questions! to monitor exported energy of a photovoltaic on-grid inverter

ovisopa, 2021-02-14:
Hello and congratulations for this awesome library!

A few hours ago I found this library and right now I purchased an SDM630 from energibutiken.se as I couldn't wait the +1 month shipping from aliexpress :D

I'm interested more in reading the exported energy, and I want to understand better **if I need something else than an HW-0519 TTL to RS485** to be able to read the exported energy from meter. And I'm more curious not about the total kwh of exported energy to national grid, but instead the "live" power . In the photo bellow, if my inverter produces more than the house consumes, **the power value will be negative ?** Or how I could find out if I'm exporting or consuming energy, if the power value won;t be negative, the PHASE ANGLE value will help? Any more info regarding this will be welcoming. 

[image]

I want to be able to control the water heater of my solar hot water tank, with an SSR relay, but to be able to do this, I need to know exactly how much surplus energy I have at any given time from the photovoltaic system, I plan to update the SSR relay power every 5 seconds, this way I want to limit the energy that's exported to national grid, I want to use as much of the produced energy I can.  

PS. Now I use a few PZEM004 sensors to monitor and send energy data to my home dashboard, but it's a little complicated when needing to measure the exported energy, as PZEM doesn't shows the direction of the flow, by default, and only on older version there is an option/hack to find out if you are exporting or importing energy from the national grid

Thank you.

reaper7, 2021-03-23:
sorry but i have no experience with energy production ...
I don't know what esp shows in this case

maxdd, 2021-04-03:
SDM630 has bidirectional measurement so yes, i dont have direct experience with it but with similar devices

reaper7, 2021-04-26:
lack of activity. closed

closed 2021-04-26 by reaper7