reaper7/SDM_Energy_Meter #53: Error code: 2

bluemanos, 2021-03-13:
For my Wemos D1 mini and HardwareSerial with default TX/RX pins I was getting a lot of errors with code 4 and 2. After changing the `RESPONSE_TIMEOUT` to 1000 and add a `delay(100)` before every read I decreased errors to just 1-3 per one read cycle (I'm reading 22 values in a cycle). Now just error code 2 is happening. What can I do next to eliminate it?

reaper7, 2021-03-15:
do you have only one modbus slave device on rs485 line or more?
For me it's a poor connection ... long lines without twisted pair cable or more than one slave device with the same slave address.

bluemanos, 2021-03-15:
You are good :)
1st yes - I was testing this on a long cable (10 m), but lines of the cable are twisted - ethernet cat 5 cable.
2nd yes - indeed I was checking a single SDM 630 energy meter with two software/hardware (one working for months on ESPEasy), the second one with your implementation at EspHome.

After all my tests I finally moved the wemos to the garage to monit a second Eastron and all error codes 2 moved away. But still, I have to use delay 100 to avoid error code 4 :/ It is strange that on a different Eastron, different rs485 modules I'm getting so many errors :/

Anyway - solution of the topic have been described, so I'm closing the issue. Thank you for your time :)

---
Dzięki za projekt, dobra robota. Powodzenia :)

closed 2021-03-15 by bluemanos