reaper7/SDM_Energy_Meter #55: SDM630 on ESP32 not working

Jimme2, 2021-03-19:
Hello and good evening,

I'm new in ESP32 and Modbus and have trouble to get the connection to my SDM630 working at all.
May I ask you to help me? Thanks in advance for your support!

My hardware is a ESP32 devkit v4 with a 485max and SDM630; as software I use PlatformIO.

What I did up to now:
I made new project in PlatformIO and included SDM and downloaded the example SDM_Live_page_ESP32_hwserial.
Then I included <index_page.h> and all necessary lib-deps like <AsyncTCP.h>,  <ESPAsyncWebServer.h>

Now I removed // of  #define USE_HARDWARESERIAL in SDM_Config_User.h and defined in Arduino.h the "NOT_A_PIN" to "2"  as I assume this is De/Re which is used in line "SDM sdm(Serial, 9600, NOT_A_PIN, SERIAL_8N1, 3, 1); of main.cpp).

I entered my wifi data, compiled the code and flashed the ESP - voila the webpage was working but of course all values are zero and Error counter was increasing. SDM READ ERR. CODE | 4 | code was shown.

Now I powered off and connected ESP with 485max like this: 
GPIO1 (TXd)==> DI
GPIO3 (RXd)==> RO
GPIO2 ==> DE+RE
3.3V ==>Vcc
Gnd ==> Gnd

Then I connected the 485max with the SDM630:
A ==> A
B ==> B
Gnd ==>G

And re-powered everything incl. SDM630. FYI: at the SDM630 no software changes were made, Modbus settings remained default (Adr 001, Baud 9.6k, Par None,Stop 1).

BUT: Now the Webpage still remained counting read errors and the error code also remaind on "4".
I checked wiring several times and read lot of issues here but finally found no solution to get it working.

I checked with Termite if the ESP sends requests and it seems to be working but somehow nothing comes back...
[image]

Please let me know if you have an idea what to check next - thanks in advance.

With best regards
Clemens

reaper7, 2021-03-23:
to completely eliminate a possible esp problem, do two things:

1. with Termite (as You do before) try to simulate reply and check if webpage shows updated VOLTAGE for request **01 04 00 00 00 02 71 cb**

set Termite as shown below:
[image]

paste this reply into termite reply line:
**0x01 0x04 0x04 0x43 0x66 0x33 0x34 0x1b 0x38**

wait when teermite show this request from esp:
[image]

and immediately send prepared earlier reply.

check on web page if voltage is updated.

if yes then

2. check with some logic tester or simple led with resistor if DE+RE is properly controlled via GPIO02 from ESP
(high state when ESP transmit request and low state when esp waiting for reply from sdm)

Jimme2, 2021-03-25:
Good evening Everybody,

Good news first: Its working now - thank you reaper7 for your support!

I followed your suggestions with Termite and the webpage was updated well - so far it seems the code was OK.

As I have no logic tester, I used the LED and got slightly confused: 
The DE/RE was pulled down well but somehow the onboard LED on the MAX485 did not flicker at all and I remembered that I read somewhere at the internet, that it should..
So I connected a new MAX485 (luckily I had 1 for spare) and now everything was working - but only as long as I do not connect 3,3V supply line to the MAX485. 
Somehow the MAX485 was supplied by RX/TX pins out of the ESP32. 
I thought to myself: anyway - it's working now and was happy...

Inspired by my success, I now wanted to read-in the voltages of  L2/L3 and modified main.cpp + index_page.h. 
I transfered --> and it was not working again 😠 

I searched again one more long evening - the reason for my late reply - and realized that the 2nd (and my last) MAX485 was killed also - damned. 
It seemed to got killed during transfering the code to the ESP32. Memo to myself: Next time I will disconnected the ESP32 from the converter...

Now I ordered a bunch of them at aliexpress (50 pcs for 12€) and another converter from Amazon for 8€ (yes, quite expensive, but I don't want to wait another 20-30 days for next attempt). 
This Amazon item is called DSD TECH SH-U12 and has auto-flowcontrol by MAX13487 Chip + 3 LEDs for TXD, RXD, Power and seems to be good quality. On Ali they are sold for ~2€ 🤦 💸   
I roll back my definition of "NOT_A_PIN" from "2" to "-1" and transfered my code (with disconnected converter). 
Connected the converter, started power supply and et voilá: it is working - Yippee!

So, from now I will transfer only with a disconnected converter - or has someone a better idea?

Thank you all, especially reaper7  - and keep this project running. Good job 👍 
Clemens

reaper7, 2021-04-26:
lack of activity. closed

closed 2021-04-26 by reaper7