reaper7/SDM_Energy_Meter #56: Multiple SDM with a single ESP32

maxdd, 2021-04-03:
Hello,
I admit i have low knowledge on how the library works but i know about master/slave Modbus concept.
I was wondering whether a "multiple sdm environment" can work on a single esp32.
The idea is to monitor multiple AC line.

Regards,

reaper7, 2021-04-03:
yes, one esp + one converter is enough 
and of course different slave address on each meter

maxdd, 2021-04-03:
where do you normally source this devices? from aliexpress it takes quite a lot.
I briefly looked through the code but haven't seen any api to set the slave address or anything else.
What do you normally use?

reaper7, 2021-04-03:
I bought one device a long time ago, I don't have the others.

I set the slave address and transmission parameters on the meter, the SDM220 has this functionality.

This lib allow only read from 3X input registers.

maxdd, 2021-04-03:
Ok, i might as well extend this library when i decide to dive in my project

reaper7, 2021-04-03:
on github you will find libraries that fully support the modbus protocol

closed 2021-07-28 by reaper7