reaper7/SDM_Energy_Meter #57: Getting SDM_ERR_WRONG_BYTES error

re3lex, 2021-04-28:
Hello

I have ESP32 that reads SDM220 via Modbus  using this library.
It worked long time without issues. Last time I update firmware of my ESP32 around December 2020 (though not sure what version of SDM library was used that time).

Today I updated SDM library version to last one and loaded new firmware to ESP32. As result I have started getting SDM_ERR_WRONG_BYTES error as result of values read.

As per comment next to erro code definition this error happens because `bytes b0,b1 or b2 wrong`, but I'm not big expert in Modbus protocol and not sure what that mean and how to fix it.

Can you please shed some light on what can cause this error and how to fix it?

reaper7, 2021-04-28:
try somehow to diagnose what response you are getting,

maybe these two issues will help you:
https://github.com/reaper7/SDM_Energy_Meter/issues/55
https://github.com/reaper7/SDM_Energy_Meter/issues/51

Remember also, that the library has evolved over time, the names that define registers have changed.

re3lex, 2021-04-28:
hi @reaper7 
Thanks for reply.

Checked those two issues, but, unfortunately, cannot find anything specific for my one.

Checked register addresses: they looks to be correct.

I was able to get response for some registers (digits before colon is index in uint8_t response array, value is following after colon):
```
Data for register address: 0 
  0:1; 1:4; 2:0; 3:0; 4:0; 5:2; 6:113; 7:203; 8:1
Data for register address: 6
  0:1; 1:4; 2:0; 3:6; 4:0; 5:2; 6:145; 7:202; 8:1
Data for register address: C
  0:1; 1:4; 2:0; 3:12; 4:0; 5:2; 6:177; 7:200; 8:1
Data for register address: 1E
  0:1; 1:4; 2:0; 3:30; 4:0; 5:2; 6:17; 7:205; 8:1
  ```

reaper7, 2021-04-28:
strange...
are you sure you have everything well set in SDM_Config_User.h

I understand you are using the hardware serial option ?  so is it uncommented? 
same for uart speed and other uart config?

re3lex, 2021-04-28:
yes, `#define USE_HARDWARESERIAL` is uncommented (otherwise build fails :) )

I see read activity on my SDM220, but result is the same: error 2.

re3lex, 2021-04-28:
I have connected ESP32 via usb2ttl converter to my PC and sent test value to emulate SDM response using hardware Serial1.
Test value is processed correctly.

Can you please suggest what can be an issue then?

re3lex, 2021-04-28:
I have 2 spare rs485-to-ttl converters. Tried both, but the result is the same.
Also I have 2 SDM220, tested on both, but no luck: b2 is zero instead of 0x04

re3lex, 2021-04-28:
Continue analyzing.

This is content of `sdmarr` array before sdmSer.write() operation:
```
[0:1, 1:4, 2:0, 3:0, 4:0, 5:2, 6:113, 7:203, 8:0 ]
```

This is what is returned by sdmSer.read() operation:
```
(0:1, 1:4, 2:0, 3:0, 4:0, 5:2, 6:113, 7:203, 8:1 }
```

It appears SDM220 is just echoing.

re3lex, 2021-04-28:
Finally managed to get it to work.
I used NO_PIN for DE_RE pin in SDM class instance setup (DE_RE physical pin sat on GND).
Setting DE_RE pin to IO pin and defining it in class instance setup fixed the problem.

re3lex, 2021-04-28:
@reaper7 thanks for your help!

last question: any idea why DE_RE pin on the GND has stopped to work correctly?

reaper7, 2021-04-29:
Personally I don't use a DE_RE pin, default value NOT_A_PIN is used.
My converter doesn't need this signal because it controls the flow by itself.
Note that **readVal** function does not clear the **sdmarr** buffer after sending request...
...so if there is no answer the buffer still contains first request.

But in your case something is received  (any loop in the converter?)
because there are no errors that should have happened sooner (if there is no reverse transmission)
like SDM_ERR_TIMEOUT or SDM_ERR_NOT_ENOUGHT_BYTES

re3lex, 2021-04-29:
Ok, thanks for reply and help.

closed 2021-04-29 by re3lex