reaper7/SDM_Energy_Meter #58: set WAITING_TURNAROUND_DELAY and RESPONSE_TIMEOUT in function call

Hard-n-Soft, 2021-11-01:
Hi,

thanks for providing the SDM630 lib. I don't want to use the SDM_Config_User.h file for configuration. Therefore it would make sense - from my point of view - to add a function to change the WAITING_TURNAROUND_DELAY and RESPONSE_TIMEOUT values while runtime and not by define for eample in setup(). Is that possible?

Best regards and thanks

reaper7, 2021-11-01:
as people say - everything is possible :)
For security such functions should also check values entered by the user (minimum and maximum values),
but what should they be? 
Believe me but you can't allow too much freedom because people try to enter strange values :) 

Are the SET functions alone enough? should GET functions also appear? 

I don't actively use this library so PR would be welcome and I don't have the option to test.

reaper7, 2021-11-01:
please check new 4 functions:
https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.h#L260

Hard-n-Soft, 2021-11-02:
Thanks, it's working like wanted. 1st test is working fine.

Best regards and thanks, Hard-n-Soft

closed 2021-11-05 by reaper7