reaper7/SDM_Energy_Meter #63: Timeouts are not overflow-safe

Flole998, 2022-04-14:
Everytime this code checks a timeout a non-overflow approach was chosen. See https://arduino.stackexchange.com/questions/12587/how-can-i-handle-the-millis-rollover for further explanations.

For example:
```
  resptime = millis() + msturnaround;

  while (sdmSer.available() < FRAMESIZE) {
    if (resptime < millis()) {
      readErr = SDM_ERR_TIMEOUT;                                                //err debug (4)
      break;
    }
    yield();
  }
```

should be

```
  resptime = millis();

  while (sdmSer.available() < FRAMESIZE) {
    if (millis() - resptime > msturnaround) {
      readErr = SDM_ERR_TIMEOUT;                                                //err debug (4)
      break;
    }
    yield();
  }
```

Same for the `SDM::flush` function.

reaper7, 2022-04-14:
@Flole998 - you're right.
I will be grateful if you can find time to prepare a PR
tnx!

reaper7, 2022-05-27:
done https://github.com/reaper7/SDM_Energy_Meter/commit/5121587c68992cdedc5fbdea232f055f5f120709

closed 2022-05-27 by reaper7