reaper7/SDM_Energy_Meter #66: Building error with the  'LED_BUILTIN' function

neoatomic, 2022-08-19:
New to ESP32/Arduino and I'm having some issues building. The "simple" example version works but the "live_esp32_hwserial" not.
It comes down to the 'LED_BUILTIN' function, that trows 4 errors and aborts the build process.
Is there an reason why this error would normally occur? Did i forget a lib or a setting?

I have a ESP32S & ESP32 Devkitv1 (ESP32-D0WDQ5 Rev 3) both have the some issue.

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None"

PATH sdm_live_page_esp32_hwserial.ino: In function 'void ledOn()':
sdm_live_page_esp32_hwserial:128:16: error: 'LED_BUILTIN' was not declared in this scope
   digitalWrite(LED_BUILTIN, HIGH);
           ^~~~~~~~~~~
Multiple libraries were found for "WiFi.h"

'LED_BUILTIN' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

reaper7, 2022-08-20:
Looks like your boards do not have LEDs and there is no definition for the corresponding pin.
Simplest way to fix this error is define this constant to some unused pin on your board
at top of sketch (similar to lines 3,4).
So, if you do not use for eg. pin number 12 then add line like this:
```
#define LED_BUILTIN 12
```

P.S.
You can check the definitions and pin assignments for the selected PCB here:
https://github.com/espressif/arduino-esp32/tree/master/variants

an example of a board that has a defined pin for the LED:
https://github.com/espressif/arduino-esp32/blob/master/variants/d1_mini32/pins_arduino.h#L9

neoatomic, 2022-08-20:
Thank you for the help, it got the building process to finish. 

I actually got it fully working now, but there was a little more to it. Will explain below to help others. :

_The board's actually did have onboard leds. Those ESP32's where cheap $4 variants from AliExpress.
I did a little searching and the correct board to select in Arduino is the "DOIT esp32 devkit v1" (nodemcu-32s seems to work also)
The 'LED_BUILTIN' errors are than also gone than during the building.
Next the boards Pin's. They have marked RX2 & TX2 and a RX0 & TX0 pin set. Advice use the RX2 & TX2, they have the port numbers RX2=16 TX2=17.
So make sure to change those in the SDM_Config_User.h & SDM sdm section of the sketch.
The TTL converter has 2 red leds that light up if the ports are working. So check those if they light-up, if the TX led works, the port config should be ok. The RX only lights-up if a valid signal is recognized.
So if a meter with ID-1 is sending valid data, if you have it set to ID-2 the led will not light-up. 
Also pay attention to the baud-rate, it needs to be 9600 as the new SDM meters are now factory set to 9600. (old meters where set to 4800?)_

closed 2022-08-20 by neoatomic