reaper7/SDM_Energy_Meter #67: Question about what RS485 converter to use

panbak, 2022-08-30:
Hello and thank you for tour work.
It is unclear for me what module I should use to make this work. Is any of the modules shown to links bellow compatible to use with your library for my ESP32  project?

[TTL to RS-485](https://www.amazon.com/YWBL-WH-RS-485-Converter-Adapter-Raspberry/dp/B07S1BN6ZK)

[UART Serial Port to RS485](https://www.amazon.com/XY-017-Serial-Converter-Automatic-Control/dp/B09L8CW98M/ref=sr_1_30?keywords=ttl+to+rs485&qid=1661840982&sr=8-30)

Thank you and sorry if my question is irrelevant with the library.

reaper7, 2022-08-30:
Hi!
The second module on your list looks similar to mine,
according to the description, it has automatic flow control...
so it can be connected with four wires(+,-.TX,RX), the first module requires five wires (+,-.TX,RX,FLOWCONTROL).

The second module also has two LEDs, helpful in diagnostics.

...but the decision is yours :)

panbak, 2022-08-30:
Thank you for the fast reply!

I know this is not stack overflow but could validate that my code looks good? I have set the correct baud rate for the ddm18sd power meter I am using and the slave id is 9. I am currently using the module with no automatic flow control and I control it from GPIO4.

`#include <SDM.h>                                                                

SDM sdm(Serial1, SDM_UART_BAUD, 4, SERIAL_8N1, SDM_RX_PIN, SDM_TX_PIN);                                                                          //for SWSERIAL

void setup() {
  Serial.begin(115200); 
  sdm.begin();
}

void loop() {
  char bufout[10];
  sprintf(bufout, "%c[1;0H", 27);
  Serial.print(bufout);

  Serial.print("Voltage:   ");
  Serial.print(sdm.readVal(DDM_PHASE_1_VOLTAGE, 0x09), 2);                            //display voltage
  Serial.println("V");

  Serial.print("Current:   ");
  Serial.print(sdm.readVal(DDM_PHASE_1_CURRENT, 0x09), 2);                            //display current
  Serial.println("A");

  Serial.print("Power:     ");
  Serial.print(sdm.readVal(DDM_PHASE_1_POWER, 0x09), 2);                              //display power
  Serial.println("W");

  Serial.print("Frequency: ");
  Serial.print(sdm.readVal(DDM_FREQUENCY, 0x09), 2);                                  //display frequency
  Serial.println("Hz");

  delay(1000);                                                                  //wait a while before next loop
}`

reaper7, 2022-08-30:
looks good...
remember to uncomment this line:
https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM_Config_User.h#L17

closed 2022-08-31 by panbak