reaper7/SDM_Energy_Meter #68: Please help with the connection

shgubar, 2022-09-07:
Hello
Please help with the connection, because I already spent a lot of time on this but nothing happened to me.
The current connection looks like the picture below (everything seems to be correct):

[image]

Here is the configuration file:

```
/*
*  define or undefine USE_HARDWARESERIAL (uncomment only one or none)
*/
//#undef USE_HARDWARESERIAL
#define USE_HARDWARESERIAL

//------------------------------------------------------------------------------

/*
*  define user baudrate
*/
#define SDM_UART_BAUD                       9600

//------------------------------------------------------------------------------

/*
*  define user SDM_RX_PIN and SDM_TX_PIN for esp/avr Software Serial option
*  or ESP32 with Hardware Serial if default core pins are not suitable
*/
#if defined ( USE_HARDWARESERIAL )
  #if defined ( ESP32 )
    #define SDM_RX_PIN                        13
    #define SDM_TX_PIN                        15
  #endif
#else
  #if defined ( ESP8266 ) || defined ( ESP32 )
    #define SDM_RX_PIN                        13
    #define SDM_TX_PIN                        15
  #else
    #define SDM_RX_PIN                        10
    #define SDM_TX_PIN                        11
  #endif
#endif

//------------------------------------------------------------------------------

/*
*  define user DERE_PIN for control MAX485 DE/RE lines (connect DE & /RE together to this pin)
*/
//#define DERE_PIN                            NOT_A_PIN

//------------------------------------------------------------------------------

#if defined ( USE_HARDWARESERIAL )

  /*
  *  define user SDM_UART_CONFIG for hardware serial
  */
  //#define SDM_UART_CONFIG                   SERIAL_8N1

  //----------------------------------------------------------------------------

  /*
  *  define user SWAPHWSERIAL, if true(1) then swap uart pins from 3/1 to 13/15 (only ESP8266)
  */
  //#define SWAPHWSERIAL                      0

#else

  /*
  *  define user SDM_UART_CONFIG for software serial
  */
  //#define SDM_UART_CONFIG                   SWSERIAL_8N1

#endif

//------------------------------------------------------------------------------

/*
*  define user WAITING_TURNAROUND_DELAY time in ms to wait for process current request
*/
//#define WAITING_TURNAROUND_DELAY            200

//------------------------------------------------------------------------------

/*
*  define user RESPONSE_TIMEOUT time in ms to wait for return response from all devices before next request
*/
//#define RESPONSE_TIMEOUT                    500

//------------------------------------------------------------------------------
```

Here is the code I am trying to use to output the data:

```
#include <SDM.h>                                                                //import SDM library

#if defined ( USE_HARDWARESERIAL )                                              //for HWSERIAL

#if defined ( ESP8266 )                                                         //for ESP8266
SDM sdm(Serial1, SDM_UART_BAUD, NOT_A_PIN, SERIAL_8N1);                                  //config SDM
#elif defined ( ESP32 )                                                         //for ESP32
SDM sdm(Serial1, SDM_UART_BAUD, NOT_A_PIN, SERIAL_8N1, SDM_RX_PIN, SDM_TX_PIN);          //config SDM
#else                                                                           //for AVR
SDM sdm(Serial1, SDM_UART_BAUD, NOT_A_PIN);                                              //config SDM on Serial1 (if available!)
#endif

#else                                                                           //for SWSERIAL

#include <SoftwareSerial.h>                                                     //import SoftwareSerial library
#if defined ( ESP8266 ) || defined ( ESP32 )                                    //for ESP
SoftwareSerial swSerSDM;                                                        //config SoftwareSerial
SDM sdm(swSerSDM, SDM_UART_BAUD, NOT_A_PIN, SWSERIAL_8N1, SDM_RX_PIN, SDM_TX_PIN);       //config SDM
#else                                                                           //for AVR
SoftwareSerial swSerSDM(SDM_RX_PIN, SDM_TX_PIN);                                //config SoftwareSerial
SDM sdm(swSerSDM, SDM_UART_BAUD, NOT_A_PIN);                                             //config SDM
#endif

#endif

void setup() {
  Serial.begin(115200);                                                         //initialize serial
  sdm.begin();                                                                  //initialize SDM communication
}

void loop() {
  char bufout[10];
  sprintf(bufout, "%c[1;0H", 27);
  Serial.print(bufout);

  Serial.print("Voltage:   ");
  Serial.print(sdm.readVal(SDM_PHASE_1_VOLTAGE), 2);                            //display voltage
  Serial.println("V");

  Serial.print("Current:   ");
  Serial.print(sdm.readVal(SDM_PHASE_1_CURRENT), 2);                            //display current
  Serial.println("A");

  Serial.print("Power:     ");
  Serial.print(sdm.readVal(SDM_PHASE_1_POWER), 2);                              //display power
  Serial.println("W");

  Serial.print("Frequency: ");
  Serial.print(sdm.readVal(SDM_FREQUENCY), 2);                                  //display frequency
  Serial.println("Hz");

  delay(1000);                                                                  //wait a while before next loop
}
```
My output in Serial Monitor:

```
16:32:55.321 -> [1;0HVoltage:   nanV
16:32:56.017 -> Current:   nanA
16:32:56.739 -> Power:     nanW
16:32:57.435 -> Frequency: nanHz
```

Thanks in advance.

FYI: Same converter I use with Raspberry Pi 4 and connect to RX_UART & TX_UART, read with phyton code - everithing work fine.

reaper7, 2022-09-07:
You are defined **Serial1** as i/o for sdm library
but your connection pins are Serial0

shgubar, 2022-09-08:
Thanks for your quick answer.

I tried, as you suggested, the piece of code below (did I understand correctly?):

```
#if defined ( ESP8266 )                                                         //for ESP8266
SDM sdm(Serial0, SDM_UART_BAUD, NOT_A_PIN, SERIAL_8N1);                                  //config SDM
#elif defined ( ESP32 )                                                         //for ESP32
SDM sdm(Serial0, SDM_UART_BAUD, NOT_A_PIN, SERIAL_8N1, SDM_RX_PIN, SDM_TX_PIN);          //config SDM
#else                                                                           //for AVR
SDM sdm(Serial0, SDM_UART_BAUD, NOT_A_PIN);                                              //config SDM on Serial1 (if available!)
#endif
```

But I immediately got an error:

```
Arduino: 1.8.13 (Mac OS X), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, OTA, Only Sketch, 921600"

sdm_simple:25:9: error: 'Serial0' was not declared in this scope
 SDM sdm(Serial0, SDM_UART_BAUD, NOT_A_PIN, SERIAL_8N1);                                  //config SDM
         ^
Using library SDM_Energy_Meter-2.2.0 at version 2.2.0 in folder: /Users/sh/Documents/Arduino/libraries/SDM_Energy_Meter-2.2.0 
exit status 1
'Serial0' was not declared in this scope
```

reaper7, 2022-09-08:
Serial not Serial0

TX LED diode on the converter blinking?

You cannot use the serial port at the same time for debugging and for communication with sdm.

shgubar, 2022-09-08:
> You cannot use the serial port at the same time for debugging and for communication with sdm.

I already tried to connect through the power supply, after downloading the sketch from the computer. But doesn't work.

> TX LED diode on the converter blinking?

Yes, blinking

> what happens with RX LED

Nothing, not blinking

shgubar, 2022-09-08:
Now I tried another sketch - sdm_live_page_esp8266_hwserial - from last release (SDM_Energy_Meter-2.2.0)

https://user-images.githubusercontent.com/75221402/189102844-ca9d49b9-1f40-45bb-ac61-9b0795cabc31.MOV

And connect through the power supply.
On web page zeroes:

 
[image]

reaper7, 2022-09-08:
4 mean timeout
https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.cpp#L112

maybe you have set different baudrate or stopbits parameters ??

shgubar, 2022-09-08:
If I use SoftwareSerial:
`SDM sdm(swSerSDM, 9600, NOT_A_PIN, SWSERIAL_8N1, 13, 15);`
All working fine

reaper7, 2022-09-08:
have you tried an older version of the library ?? 2.1.1
in this new one, timeouts were modified, but unfortunately without verification (I don't have these devices anymore)

reaper7, 2022-09-08:
if swserial works and hw doesn't, maybe you swapped TX / RX pins ?? wemos often had swapped overprints...
or something is plugged into wemos usb port and is interfering with transmission

shgubar, 2022-09-08:
> if swserial works and hw doesn't, maybe you swapped TX / RX pins ?? wemos often had swapped overprints... or something is plugged into wemos usb port and is interfering with transmission

I have already tried to change it, but it does not help

shgubar, 2022-09-08:
> have you tried an older version of the library ?? 2.1.1 in this new one, timeouts were modified, but unfortunately without verification (I don't have these devices anymore)

It's all the same

shgubar, 2022-09-08:
What was done:
1) Used external power supply
2) Changed RX/TX

[image]

closed 2024-03-07 by shgubar