reaper7/SDM_Energy_Meter #71: Example Sketch for Mulitple SDM630 (3Pc.) ID1 to ID3 -> Output Influx DB

fliegenderfish, 2023-01-13:
Hello,
I Just tried to find out how I can Collect the Data from 3 different SDM630 on one MODBUS....

The Examples show only one SDM at the time, an in the Readme the Ex. for multiple Devices does not fit the examples.
So could you give an idea?

The goal would be to the data from 3 SDM 630 into the a InfluxDB or via MQTT.
Thanks a lot!

reaper7, 2023-01-13:
I don't have any example, it's hard for me to talk about influxdb as well (I'm not the author of the database example).

With a few meters, I suggest creating a 2d array ... for e.g. https://www.tutorialspoint.com/arduino/arduino_multi_dimensional_arrays.htm
first in the loop you read the meters into the array
and then in another loop you send this data (e.g. to mqtt)

fliegenderfish, 2023-01-13:
Yes you are right, the export of the collected data is the second step.... But I already struggle with collecting the data of multiple SDMs and display it for test example "Simple live Page"...

Since in the SDM.h you can only define "one" NODE, 0x01 or 0x02 or 0x03....
[image]

and collect this "one" NODE:
[image]

And I have no Idea how to implement the described collection of multiple devices in the existing files:
[image]

Thanks very much!

reaper7, 2023-01-13:
SDM.h, in this case, specifies the default value (if you don't specify parameters).

so...
this command will read the meter with id set to 1 (because by default we read it if the user does not provide an id) :
float power1 = sdm.readVal(SDM_PHASE_1_POWER);

this command will do exactly the same, it will read the meter with id 1, but explicitly indicated by the user :
float power1 = sdm.readVal(SDM_PHASE_1_POWER, 0x01);

So to read phase 1 power from consecutive meters with ids 1 to 3, you can do it like this:
```
  for (uint8_t meter_id = 1; meter_id < 3; meter_id++) {
    float power = sdm.readVal(SDM_PHASE_1_POWER, meter_id);

    //do something with this value
  }
```

shgubar, 2023-04-25:
Hi.
What about(?):
```
sdm.getSuccCount();
sdm.getErrCount();
sdm.getErrCode();
```

Use it or something else(?):
```
sdm.getSuccCount(0x01);
sdm.getErrCount(0x01);
sdm.getErrCode(0x01);
```

reaper7, 2023-04-25:
@shgubar - problems section:  https://github.com/reaper7/SDM_Energy_Meter#problems

shgubar, 2023-04-26:
Yes, thank you.  That is, there is no provision to show errors for each one separately?  Did I understand correctly?

reaper7, 2023-04-26:
Exactly, these functions return a global values (not individually per meter)

closed 2023-10-03 by reaper7