reaper7/SDM_Energy_Meter #76: Error: SoftwareSerial::availableForWrite()' marked override

jnissen, 2023-04-04:
Hello,
Iám use the Simple Expample. On Compiling I get anthe following Error.

.platformio\packages\framework-arduinoespressif8266@2.7.4+9\libraries\SoftwareSerial\src\SoftwareSerial.h:132:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override
     int availableForWrite() override {
Can anybody Help ?

reaper7, 2023-04-04:
Try to use older branch, for e.g. https://github.com/reaper7/SDM_Energy_Meter/tree/old_esp_swserial_600_701
Current version this lib requires SoftwareSerial >= v8.0.1

jnissen, 2023-04-04:
Hello,
i have downloaded old Version. 
*  2016-2022 Reaper7 (tested on wemos d1 mini->ESP8266 with Arduino 1.8.10 & 2.5.2 esp8266 core)
But Same error
If I Compare Source auf sdm.cpp, sdm.h, sdm_config-User.h there only Changed is the Version Line above.

Do You mean, i have to use an other version of SoftwareSerial.h ?

reaper7, 2023-04-04:
This is problem with SoftwareSerial not this lib. Try to update sw lib

jnissen, 2023-04-04:
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps =
 plerup/EspSoftwareSerial @ ^6.0.0

ompiling .pio\build\nodemcuv2\FrameworkArduino\MD5Builder.cpp.o
In file included from .pio\libdeps\nodemcuv2\EspSoftwareSerial\src\SoftwareSerial.cpp:23:0:
.pio\libdeps\nodemcuv2\EspSoftwareSerial\src\SoftwareSerial.h:136:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override
     int availableForWrite() override {
         ^
.pio\libdeps\nodemcuv2\EspSoftwareSerial\src\SoftwareSerial.h:163:9: error: 'int SoftwareSerial::read(uint8_t*, size_t)' marked override, but does not override

reaper7, 2023-04-04:
Sorry but I didn't check it on platformio, it's hard for me to say what needs to be changed.
Try a newer version of the library, for e.g.:
```
lib_deps =
plerup/EspSoftwareSerial @ ^7.0.0
```

reaper7, 2023-04-04:
looks like this ini works:
```
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps =
  plerup/EspSoftwareSerial @ ^8.0.1
platform_packages = 
  framework-arduinoespressif8266@https://github.com/esp8266/Arduino.git
  mcspr/toolchain-xtensa@^5.100200.0
```

I don't know if it works but it compiles without errors

jnissen, 2023-04-05:
Hello,
thanks for your reply..
I get an new error 

c:/users/XXXXXX/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\main.cpp.o:(.text.setup+0x8): undefined reference to `_ZN3SDM5beginEv'
c:/users/XXXXXX/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\main.cpp.o:(.text.loop+0x24): undefined reference to `_ZN3SDM7readValEth'
c:/users/XXXXXX/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\main.cpp.o:(.text.loop+0x5a): undefined reference to `_ZN3SDM7readValEth'
c:/users/XXXXXX/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\main.cpp.o:(.text.loop+0x87): undefined reference to `_ZN3SDM7readValEth'
c:/users/XXXXXX/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\main.cpp.o: in function `loop':
main.cpp:(.text.loop+0xb4): undefined reference to `_ZN3SDM7readValEth'
c:/users/XXXXXX/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: main.cpp:(.text.loop+0xe3): undefined reference to `_ZN3SDM7readValEth'
c:/users/XXXXXX/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\main.cpp.o:(.text.startup._GLOBAL__sub_I_swSerSDM+0x4): undefined reference to `_ZN3SDMC1ERN17EspSoftwareSerial9BasicUARTINS0_16GpioCapabilitiesEEEliiaa'
c:/users/XXXXXX/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\main.cpp.o: in function `_GLOBAL__sub_I_swSerSDM':
main.cpp:(.text.startup._GLOBAL__sub_I_swSerSDM+0x32): undefined reference to `_ZN3SDMC1ERN17EspSoftwareSerial9BasicUARTINS0_16GpioCapabilitiesEEEliiaa'
c:/users/XXXXXX/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\main.cpp.o:(.text.exit._GLOBAL__sub_D_swSerSDM+0x0): undefined reference to `_ZN3SDMD1Ev' 
c:/users/XXXXXX/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\main.cpp.o: in function `_GLOBAL__sub_D_swSerSDM':
main.cpp:(.text.exit._GLOBAL__sub_D_swSerSDM+0x10): undefined reference to `_ZN3SDMD1Ev'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nodemcuv2\firmware.elf] Error 1
=============================================== [FAILED] Took 64.17 seconds

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps =
  plerup/EspSoftwareSerial @ ^8.0.1
platform_packages = 
  framework-arduinoespressif8266@https://github.com/esp8266/Arduino.git
  mcspr/toolchain-xtensa@^5.100200.0

reaper7, 2023-04-05:
I can't help, I'm not a platformio expert.
I suggest clearing the project cache, remove old toolchains.
...or just compile the project on arduino ;)

jnissen, 2023-04-05:
Ok, I am test it on Arduino... :-)

jnissen, 2023-04-05:
Wich RS485 Converter do you use for HWSerial ?

reaper7, 2023-04-05:
as in the pictures on the main page, with automatic flow control,
something like this:
https://pl.aliexpress.com/item/33054008216.html
https://pl.aliexpress.com/item/1005004379628378.html

closed 2023-10-03 by reaper7