reaper7/SDM_Energy_Meter #77: What to do to add TAC1100 meter?

gibzwein, 2023-09-07:
Hi,
What should I do to add TAC1100 meter?
Best regards
[User Guide for TAC1100 Series V1.1.pdf](https://github.com/reaper7/SDM_Energy_Meter/files/12552527/User.Guide.for.TAC1100.Series.V1.1.pdf)
[TAC1100series Modbus Protocol v1.0.pdf](https://github.com/reaper7/SDM_Energy_Meter/files/12552528/TAC1100series.Modbus.Protocol.v1.0.pdf)

hosek, 2024-06-06:
Hi, Im trying it now, seems thats supported out of the box:
https://github.com/PaulusElektrus/sdm120_esp32_influxdb/tree/master

gibzwein, 2024-06-10:
> Hi, Im trying it now, seems thats supported out of the box: https://github.com/PaulusElektrus/sdm120_esp32_influxdb/tree/master

TAC has a bit different registers than SDM, but it's easy to fix.